moggieuk/Happy-Hare #111: Bug sync to extruder all of the sudden not working while printing on 2.3

rwijnhov, 2023-12-16:
All of the sudden sync to extruder didn't work during a print. firmware 2.3 sync to extruder is enabled in the parameters. I atached mmu log.

[mmu.zip](https://github.com/moggieuk/Happy-Hare/files/13692586/mmu.zip)

rwijnhov, 2023-12-16:
Printer restart made it working again. So i guess there is somewhere a bug that triggers it to not turn on.

moggieuk, 2023-12-18:
Happy to look at it but can you narrow down the problem or at least the part of the log that shows this not happening?  I took a quick look and other than some probably "too fast purging" during your print start I'm not seeing it.

strueman, 2023-12-28:
Sync is doing odd things in this latest update. It disengages after tool change, after print start macro etc. It seems to disable anytime there is an extruder reset in the gcode or a macro.

moggieuk, 2024-01-09:
Unless I hear back with a log I'm going to close out this issue.

rwijnhov, 2024-11-29:
For me sync isnt working at all with the latest release

rwijnhov, 2024-11-29:
I will do some log check the coming week and update this topic.

rwijnhov, 2025-01-05:
It is fixed I needed to up the runcurrent for my gear motor from 0.6 to 0.8 (nema 17 motor) now it works again.

closed 2025-01-05 by rwijnhov