moggieuk/Happy-Hare #117: What controls the movement speed to and from the cleaning tower?

erazor666, 2023-12-18:
Not an issue per say but i’m having trouble finding the setting for this, it moves so slowly that ooze has time to come out and solidify even. I’ve seen it asked a few times on discord with no proper answer either.

moggieuk, 2023-12-18:
Slow speed happens if you move to somewhere else (in your own macro) and don't return the toolhead to where it was left by HH.  It WILL go back to the correct position but does it slowly.  It can't be sped up because it is only expecting Z_hop moves and if it is sped up that will fail...

Best pattern is:
save toolhead position at start of you macro / logic
do movement
restore toolhead position as end of your macro / logic at any speed you want.

E.g.
If you create a _MMU_POST_LOAD macro that might wipe the nozzle and then restore back to the position saved at the start of _MMU_PRE_UNLOAD...

So: SAVE_GCODE_STATE NAME=MY_state in _MMU_PRE_UNLOAD
And: RESTORE_GCODE_STATE NAME=MY_state MOVE_SPEED=<fast> MOVE=1 in _MMU_POST_LOAD

Meaning restore to where HH left the toolhead.

Happy Hare uses a conservative movement speed because it only expects and controls z_hop moves. 

Maybe this should be exposed so I'll mark this as a feature.

moggieuk, 2026-10-21:
Thanks, and sorry this stayed open so long. This is now exposed in Happy Hare v4 (v4.1 is the current release), so I'm closing it.

All parking and return-to-print movement in v4 is handled by the sequence macros, and the speeds are configurable:
- `variable_park_travel_speed`: X/Y travel, including the move back to the print
- `variable_park_lift_speed`: Z-only moves

Both are in `_MMU_SEQUENCE_VARS` in `mmu_macro_vars.cfg`, or in menuconfig under "Macro Variables" -> "Sequence macros" -> "Movement speeds". Happy Hare's final "make sure we're back where we started" move uses the same travel speed, so it's no longer a slow crawl back to the print.

If you want to wipe the nozzle or visit a cleaning station after a load, add it with `variable_user_post_load_extension`. It runs before Happy Hare restores the toolhead position, so you don't need your own save/restore any more.

If there's something else v4 doesn't do here, please open a new feature request against v4, which is being actively worked on.

closed 2026-10-21 by moggieuk