moggieuk/Happy-Hare #118: Error after clean install "Unable to parse option 'happy_hare_version' in section 'mmu'"

DarkSoulRL, 2023-12-18:
how to repeat the error
./install.sh -i
after restarting in klipper I see a red window with an error
"Unable to parse option 'happy_hare_version' in section 'mmu'"
mmu_parameters.cfg looks like this:
```
[mmu]
happy_hare_version: {happy_hare_version}	# Don't mess, used for upgrade detection
```

jalexandre0, 2023-12-18:
Quick fix: 

Change that line to
happy_hare_version: 2.3 # Don't mess, used for upgrade detection

moggieuk, 2023-12-18:
This problem was fixed.  Manually setting the version to 2.3 is fine.  Also, force update-manager to refresh and double check you have the latest version (no need to run ./install.sh again -- only do that when it instructs you too.

closed 2024-02-05 by moggieuk