moggieuk/Happy-Hare #119: filametrix: filament loading does not work correctly

DarkSoulRL, 2023-12-18:
Initially it was configured without using filametrix. The filament was loaded correctly, up to the nozzle.
Then I connected mmu_filametrix.cfg. My settings are all default except variable_pin_loc position

Then I connected mmu_filametrix.cfg. My settings are all default except variable_pin_loc position

In mmu_parameters.cfg:
form_tip_macro: _MMU_CUT_TIP
After the extuder there is a sensor, it is connected.
toolhead_sensor_to_nozzle: 62

The problem is that after trimming and reload filament, a lot of plastic is squeezed out of the nozzle, and I can't adjust it at all.
I tried to manually enter the output_park_pos value in the mmu_filametrix.cfg file and it did not have any effect.

`23:25:47 
// MMU [T2] >>> [En] >...... [Ex] .. [Ts] .. [Nz] 44.6mm (e:44.6mm)
23:25:47 
// - DEBUG: Loading bowden tube
23:25:47 
// - - - STEPPER: GEAR: dist=997.1, speed=160.0, accel=400.0, sync=False, wait=False
23:25:53 
// - - TRACE: Course loading move into bowden. Stepper: 'gear' moved 997.1mm, encoder measured 998.8mm (delta -1.7mm). Pos: @1041.7, (1043.4mm)
23:25:53 
// MMU [T2] >>> [En] >>>>>>> [Ex] .. [Ts] .. [Nz] 1041.7mm (e:1043.4mm)
23:25:53 
// - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Loading Ext' OLD_ACTION='Loading'
23:25:53 
// - DEBUG: Loading filament into extruder
23:25:53 
// - - TRACE: _ensure_safe_extruder_temperature: current_temp=200.03, paused_extruder_temp=None, current_target_temp=200.0, klipper_minimum_temp=170.0, default_extruder_temp=200.0, source=auto
23:25:53 
// - DEBUG: Homing up to 40.0mm to toolhead sensor (synced)
23:25:54 
// - - - STEPPER: GEAR+EXTRUDER HOME: dist=40.0, speed=20.0, accel=400.0, endstop_name=mmu_toolhead, sync=False, wait=False
23:25:56 
// - - TRACE: Homing to toolhead sensor. Stepper: 'gear+extruder' homed after moving 38.1mm (of max 40.0mm), encoder measured 38.4mm (delta -0.2mm). Pos: @1079.8, (1081.8mm)
23:25:56 
// MMU [T2] >>> [En] >>>>>>> [Ex] >| [Ts] .. [Nz] 1079.8mm (e:1081.8mm)
23:25:56 
// - DEBUG: Loading last 62.0mm to the nozzle...
23:25:56 
// - - - STEPPER: GEAR+EXTRUDER: dist=62.0, speed=18.0, accel=400.0, sync=False, wait=True
23:26:00 
// - - TRACE: Loading filament to nozzle. Stepper: 'gear+extruder' moved 62.0mm, encoder measured 60.7mm (delta 1.3mm). Pos: @1141.8, (1142.5mm)
23:26:00 
// - DEBUG: Total measured movement: 60.7mm, total delta: 1.3mm
23:26:00 
// MMU [T2] >>> [En] >>>>>>> [Ex] >> [Ts] >> [Nz] LOADED 1141.8mm (e:1142.5mm)
23:26:00 
// - DEBUG: Filament should loaded to nozzle
23:26:00 
// - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Loading' OLD_ACTION='Loading Ext'
23:26:00 
// Load of 1141.8mm filament successful (encoder measured 1143.2mm)`

moggieuk, 2023-12-19:
I'm not sure what the problem is give that your comment is just "a lot of plastic came out of the nozzle".

If you use the Filametrix style filament cutter, some plastic will be left in the nozzle.  The amount is adjustable by you in the variable settings for `_MMU_CUT_TIP`.   This "residual plastic" is compensated for by shortening the next load that Happy Hare does.  However, if you reset the machine or MMU then knowledge of this residual plastic and the FIRST load will "overload" a little until Happy Hare catches up with state.

DarkSoulRL, 2023-12-19:
No, something isn't working right here. 
With each change of filament, approximately 30mm of excess is squeezed out. It's a lot and the head gets stuck in it.
Looking at the log that I attached above, from the toolhead sensor to the nozzle it ALWAYS squeezes out 62mm
`23:25:56  // - DEBUG: Loading last 62.0mm to the nozzle... `
The correct size after cutting should only be 20 mm, which I checked by making direct changes to mmu.py at line 3282
length = 20

That is, it is not considered that the filament is cut by a knife at a given distance, and it extrudes 62mm, which is specified in the toolhead_sensor_to_nozzle setting: 62

DarkSoulRL, 2023-12-19:
I think I found what the problem is. My extruder cannot move the filament at a speed higher than 30mm/s. And in the retract setting before trimming, the speed is 50 mm/s. Now I’ll figure it out and be sure to write what happened in the end.

DarkSoulRL, 2023-12-19:
This slightly reduced the amount of plastic extruded when changing filament, but did not solve the problem, since it still extrudes 62mm, which is the distance from the toolhead_sensor to the nozzle.

And the position of the knife for cutting and the size of the retract must also be calculated.

moggieuk, 2023-12-19:
Ah, thanks for updating this Issue.  Yes, moving too fast is not possible to detect and extruders vary so widely in there movement speed.  Since HH "measures" the movement reported via klipper it is impossible to detect skipped steps.

I have not just tweaked the default speeds and probably should reduce them.  I assume you are using a SB with CW2 extruder?  What stepper are you using?

moggieuk, 2023-12-19:
The position the filament is left in the extruder is reported by the macro as `output_park_pos`.  This value is written to `mmu.log`.  I would suggest you look at the log to see what the value is (it is calculated based on the configuration of the macro).  Then HH should reduce the remaining extruder movement by this value (and add +10mm safety margin)..

What are your macro settings?  Can you attached `mmu_filametrix.cfg`?

bartoszwas, 2023-12-19:
Filametrix is quite tricky to configure...

What I did - in slicer cooling tube position to 0, cooling tube length to 0, filament parking position to 0, extra loading distance to 6mm

In klipper I commented out HH's _MMU_CUT_TIP macro and used one provided in filametrix, and tune toolhead_extruder_to_nozzle variable in emu_parameters.cfg. Also I had to do this: toolhead_ignore_load_error: 1 and variable_output_park_pos: 20 in filametrix cfg otherwise I had error "filament seems to be stuck in extruder". 

Im not sure is this correct but it works.

@moggieuk - I know this issue is not for this question bu I'll ask. Is there option to make servo arm active up for Triple Decky three position top hats? Mine goes up for short time then goes back to move position. I have changed from standard Sturdy Bunny blocks to Triple Decky and added "t" suffix in config. Maybe I should install HH again?

DarkSoulRL, 2023-12-19:
I use extruder SB, stepper MOONS NEMA14 Motor 8T. I solved the problem with skipping steps by disabling stealthchop. Now a speed of 50 mm/s does not cause problems.
[config.zip](https://github.com/moggieuk/Happy-Hare/files/13720499/config.zip)
[image]

[bartoszwas](https://github.com/bartoszwas), where u commented _MMU_CUT_TIP?

moggieuk, 2023-12-20:
> I use extruder SB, stepper MOONS NEMA14 Motor 8T. I solved the problem with skipping steps by disabling stealthchop. Now a speed of 50 mm/s does not cause problems. [...]

Yes, stealthchop on the extruder does not make sense... it limits speed and greatly reduces torque...  Always go with spreadcycle.

moggieuk, 2023-12-20:
> Filametrix is quite tricky to configure...
>
> What I did - in slicer cooling tube position to 0, cooling tube length to 0, filament parking position to 0, extra loading distance to 6mm
>
> In klipper I [...]

Not sure I understand. The "up" position is to release the filament in both he TB and TD designs (ThumperBlocks is a modification of TripleDecky).  The "move" position is the neutral used when selector wants to move..

BTW your _MMU_CUT_TIP changes mean that something isn't correct.  You shouldn't really be ignoring load errors - it means a config setting isn't correct...

bartoszwas, 2023-12-20:
My bad. I've set  toolhead_ignore_load_error: 1 because I didn't retract after cut so loading distnance was very small like 15mm.

According to servo. Mine servo is not holding in up position. When it moves to next block it should go to up and hold in up but it doesn't so filament trap is not released...

DarkSoulRL, 2023-12-21:
I still haven’t figured out how to correctly adjust the amount of plastic extruded after loading the filament?

moggieuk, 2023-12-21:
> My bad. I've set toolhead_ignore_load_error: 1 because I didn't retract after cut so loading distnance was very small like 15mm.
>
> According to servo. Mine servo is not holding in up position. When [...]

The servo will be in the "move" position when the selector moves.  With 3 position servo it is therefore necessary to define three positions: (your angles will vary)
```
servo_up_angle: 125			# Default: MG90S servo: Up=30    ; SAVOX SH0255MG: Up=140
servo_down_angle: 45			# Default: MG90S servo: Down=140 ; SAVOX SH0255MG: Down=30
servo_move_angle: 110			# Optional angle used when selector is moved (defaults to up position)
```

closed 2024-01-22 by DarkSoulRL