moggieuk/Happy-Hare #121: Error after update: No module named 'extras.led_effect'

berreka, 2023-12-21:
happy-hare version v2.3.1-3-gcec9e661
klippy.log shows:

```
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 175, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 141, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/pi/klipper/klippy/klippy.py", line 121, in load_object
    mod = importlib.import_module('extras.' + module_name)
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/pi/klipper/klippy/extras/mmu.py", line 19, in <module>
    from extras.mmu_led_effect import MmuLedEffect
  File "/home/pi/klipper/klippy/extras/mmu_led_effect.py", line 15, in <module>
    from extras.led_effect import ledEffect
ModuleNotFoundError: No module named 'extras.led_effect'
webhooks client 4134887800: New connection
webhooks client 4134887800: Client info {'program': 'Moonraker', 'version': 'v0.8.0-209-g4235789'}
```
Great work

berreka, 2023-12-21:
I've been able to workarround this issue installing julianschill/klipper-led_effect/ and coping klipper-led_effect/sre/led_effect.py to Happy-hare/extra/led_effect.py

berreka, 2023-12-22:
happy-hare v2.3.1-5-g0a46ab73 solved this issue for me

moggieuk, 2023-12-22:
I rewrote the way the `mmu_led_effects` wrapper worked so that led-effects is optional (although you still can't use the LED effects without out).

halofx, 2023-12-22:
I just updated to v2.3.1-5-g0a46ab73, and receive the error  `No module named 'extras.mmu_led_effect'`
Is this the same issue, just kicked from line 15 to 19?

My log shows:
```
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/biqu/klipper/klippy/klippy.py", line 175, in _connect
    self._read_config()
  File "/home/biqu/klipper/klippy/klippy.py", line 141, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/biqu/klipper/klippy/klippy.py", line 121, in load_object
    mod = importlib.import_module('extras.' + module_name)
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/biqu/klipper/klippy/extras/mmu.py", line 19, in <module>
    from extras.mmu_led_effect import MmuLedEffect
ModuleNotFoundError: No module named 'extras.mmu_led_effect'
webhooks client 281473744697424: New connection
webhooks client 281473744697424: Client info {'program': 'Moonraker', 'version': 'v0.8.0-220-gaa16dd6'}
```

halofx, 2023-12-22:
Looks like the issue is a missing symlink ~/Happy-Hare/extras/mmu_led_effect.py ~/klipper/klippy/xtras/mmu_led_effect.py

Adding this link got me back up and going.

seacred, 2024-01-14:
i'm currently on v2.4.0-8-g507f1648 and the issue persists
[image]

symlink did not work neither copying klipper-led_effect/src/led_effect.py to Happy-hare/extras/led_effect.py

florin03, 2024-01-15:
cd ~/Happy-Hare
./install.sh

it will fix the error.

moggieuk, 2024-02-05:
Closing issue because this was just a setup issue

closed 2024-02-05 by moggieuk