moggieuk/Happy-Hare #122: An Upgrade routine is required

lettore, 2023-12-22:
Upgrading happy hare is really a pain, with every new upgrades something is not working, now it complains about some LEDs. I don't even understand why if some functionality is added it should change the config reference that worked until the previous version.
I kindly ask you to introduce a upgrade.sh that translates a working config from a specific version to the new one. For example asking "you need LED support?" no as I don't have LEDs.
If it's not possible it should be present a guide, step by step, to how fix the config.

salocain, 2023-12-22:
yes i agree with that , too many changes , and alway redo a config is a pain...

moggieuk, 2024-01-08:
Well, I try very hard to make the install.sh perform and upgrade to existing configuration.  The installer is a HUGE time sink because of the difficulties in testing and the almost infinite starting points.  Would love some help on this....

moggieuk, 2024-02-05:
Closing because this is old and HH doesn't use manual steppers anymore

closed 2024-02-05 by moggieuk