moggieuk/Happy-Hare #123: servo problem

salocain, 2023-12-22:
hello,

i don't know if it's happy hare problem or not , but after update my servo go to disfonctioning, i replace it , i find a bad connection , i plug directly on a board (fytsec) but now , servo didn't work's.

i test differents servo (sg90) and same result , i hear a little noise wen i push up/down command but.....

it's a possible coincidence ,but....not sure
another possibility is my board is dead , but ...why, everithing working before...

pulse config or another thing ?

salocain, 2023-12-23:
i found the problem,

after try many things on the original port (gpio23 on erb board) (try change config /servo/etc)
i try on the rgb port with gpio21 --> i modify the gpio parametre and work 

the question is WHY my GPIO23 is DEAD  ??

berreka, 2023-12-23:
Checking the schematics, the servo VCC fuse on the ERB board might be blown, it's labeled F2 here:

https://github.com/FYSETC/FYSETC-ERB/blob/main/hardware/Silk%20Fixed.pdf

moggieuk, 2024-01-09:
Thanks for documenting here.  There is nothing an update can do to trigger this.  It just sounds like an unfortunate h/w issue.  These external MCU's aren't the most robust pieces of electronics and this can happen.  The servo can tax the +5v supply too.

closed 2024-02-05 by moggieuk