moggieuk/Happy-Hare #124: Endless spool does not work when using filametrix

DarkSoulRL, 2023-12-22:
Endless spool does not work when using filametrix. 

The cut filament is not long enough for the ERCF gears to grab it and pull it out. As a result, it does not pull out the filament and blocks the selector.
Printing stops with the error:
[image]

DarkSoulRL, 2023-12-22:
Is it possible to make sure that the filament is not removed, but simply the selector goes to the next gate and pushes a new filament?

DarkSoulRL, 2023-12-24:
I found the problem
[image]

It seems to me that it would be worth limiting the autodetection of mmu_calibration_clog_length to some more real value, so that this would not happen. 7mm works great for me.

moggieuk, 2024-01-09:
This may be hitting the limitation of the ERCF design.  There are some things you can do to make it work:
1. Shorten the clog detection length.  8-10mm should be possible.  This will result in earlier detection
2. [not ERCF, but MMU's with mmu_gate sensor would be able to detect much quicker than encoder]
3. Don't leave so much filament in the extruder.  You can turn filametrix to do an initial retraction.  Then you need only leave about 5mm of filament in the extruder.  Then you are liekly to have sufficent filament to push back.
4. With the new pre-gate sensor feature, runout can be triggered much earlier without issues with the gear picking up the filament or the filament catching when being fed backwards.

moggieuk, 2024-03-11:
Reducing clog length worked, so closing this issue

closed 2024-03-11 by moggieuk

DarkSoulRL, 2024-03-11:
I no longer had this problem after upgrading to ERCF v2 + ERCT