moggieuk/Happy-Hare #125: Malformed command 'SAVE_VARIABLE VARIABLE=mmu_state_gate_material

Raabi91, 2023-12-24:
since the last update i always get the error when starting klipper

Malformed command 'SAVE_VARIABLE VARIABLE=mmu_state_gate_material VALUE='["'ASA'", "'ABS+'", "'PETG'", "'PETG'", "''", "'ABS Ca'"]''

the problem is the "ABS Ca" which i have actually set to abs. if i set it to abs again. the command goes but after a firmware restart it is automatically set to "ABS Ca" again

moggieuk, 2024-01-09:
Somehow, it has been persisted incorrectly.   If you edit `mmu_vars.cfg` you can fix and not have it reset.

Couple of questions:
1. Do yu have spoolman integration enabled?
2. Can you attach your `mmu_vars.cfg` file.  It's in printer_data/config/mmu/ directory.

Raabi91, 2024-01-10:
Problem is gone since the last updates. now everything works again as before

Spoolman is active yes 

in the mm_vars.cfg the values were also like this at this time now is it normal

this was the old cfg

`[Variables]
cutter = 0
mmu_calibration_0 = 1.0
mmu_calibration_1 = 1.065469
mmu_calibration_2 = 1.063757
mmu_calibration_3 = 1.069748
mmu_calibration_4 = 1.068892
mmu_calibration_5 = 1.068036
mmu_calibration_bowden_length = 1190.9
mmu_calibration_clog_length = 25.9
mmu_encoder_resolution = 1.026958
mmu_gear_rotation_distance = 22.997141
mmu_selector_bypass = 131.4
mmu_selector_offsets = [2.4, 25.6, 48.8, 72.0, 94.4, 117.6]
mmu_state_enable_endless_spool = 0
mmu_state_endless_spool_groups = [0, 1, 2, 2, 1, 1]
mmu_state_filament_pos = 0
mmu_state_gate_color = ['gray', '51AF34', 'black', '8ff0a4', 'white', 'FAF5F5']
mmu_state_gate_material = ['ASA', 'ABS+', 'PETG', 'PETG', '', 'ABS Ca']
mmu_state_gate_selected = 2
mmu_state_gate_spool_id = [19, 2, 16, 20, 18, 3]
mmu_state_gate_status = [1, 2, 2, 1, 1, 2]
mmu_state_tool_selected = 2
mmu_state_tool_to_gate_map = [0, 1, 2, 3, 4, 5]
mmu_statistics_gate_0 = {'pauses': 5, 'loads': 0, 'load_distance': 100849.095, 'load_delta': 2703.368, 'unloads': 0, 'unload_distance': 79816.1, 'unload_delta': 3116.335, 'servo_retries': 2, 'load_failures': 9, 'unload_failures': 2, 'quality': 0.9911130928772384}
mmu_statistics_gate_1 = {'pauses': 10, 'loads': 0, 'load_distance': 115039.012, 'load_delta': 924.053, 'unloads': 0, 'unload_distance': 52342.7, 'unload_delta': 1368.586, 'servo_retries': 1, 'load_failures': 3, 'unload_failures': 3, 'quality': 0.961726219339819}
mmu_statistics_gate_2 = {'pauses': 0, 'loads': 0, 'load_distance': 31174.071, 'load_delta': 1158.628, 'unloads': 0, 'unload_distance': 11869.8, 'unload_delta': 1474.656, 'servo_retries': 1, 'load_failures': 3, 'unload_failures': 0, 'quality': 0.9344501083336636}
mmu_statistics_gate_3 = {'pauses': 0, 'loads': 0, 'load_distance': 10877.451, 'load_delta': 48.784, 'unloads': 0, 'unload_distance': 7521.2, 'unload_delta': 8.858, 'servo_retries': 2, 'load_failures': 3, 'unload_failures': 1, 'quality': 0.9987922744299697}
mmu_statistics_gate_4 = {'pauses': 1, 'loads': 0, 'load_distance': 29513.809, 'load_delta': 418.537, 'unloads': 0, 'unload_distance': 21411.3, 'unload_delta': 1227.708, 'servo_retries': 2, 'load_failures': 2, 'unload_failures': 1, 'quality': 0.9945622210985621}
mmu_statistics_gate_5 = {'pauses': 11, 'loads': 0, 'load_distance': 109699.587, 'load_delta': 10446.962, 'unloads': 0, 'unload_distance': 59763.3, 'unload_delta': 4306.061, 'servo_retries': 5, 'load_failures': 16, 'unload_failures': 5, 'quality': 0.9797991280782012}
mmu_statistics_swaps = {'total_swaps': 330, 'time_spent_loading': 5485.01, 'time_spent_unloading': 5114.77, 'total_pauses': 27, 'time_spent_paused': 63905.53}`

The only problem was that no matter what I changed at the time it was not saved and I never entered ABS Ca, only ABS+ at the time. 
Now it says abs+ again without me having changed anything.

what i found out is that the error only came because of the space in "'ABS Ca'".

Translated with DeepL.com (free version)

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

You mentioned it went away, and in v4 gate materials are no longer passed through the SAVE_VARIABLE command line, so names with spaces are saved safely.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk