moggieuk/Happy-Hare #129: _MMU_STEP_UNLOAD_ENCODER not work

Tutankamon00, 2023-12-28:
i try to creatte my unload macro but after bowden unload i have to run _MMU_STEP_UNLOAD_ENCODER and it not work:
[image]

moggieuk, 2024-01-08:
Ah, it is called `_MMU_STEP_UNLOAD_GATE`.   Does the same thing but works both with encoder or a gate_switch (like Tradrack design)

Tutankamon00, 2024-01-09:
> Ah, it is called `_MMU_STEP_UNLOAD_GATE`. Does the same thing but works both with encoder or a gate_switch (like Tradrack design)
> 
> Do I have a bad reference to this in the doc?

[image]

moggieuk, 2024-02-05:
Doc has been fixed (finally!)

closed 2024-02-05 by moggieuk