moggieuk/Happy-Hare #131: Section 'menu __main __mmu' is not a valid config section

marcus360, 2023-12-30:
hi i get when trying to install happy hare with klipper screen this error : 

Section 'menu __main __mmu' is not a valid config section

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

i am giving you my klipper log and sorry but i didnt know where is this problem
[klippy (6).log](https://github.com/moggieuk/Happy-Hare/files/13799316/klippy.6.log)

marcus360, 2023-12-30:
for future guys, if you dont have mini display dont install library to it :)

closed 2023-12-30 by marcus360

Matze9876, 2025-02-05:
Hallo Marcus,

ich habe genau das gleiche Problem.
Wie konntest du es lösen?

Dank im Voraus
Gruß Matze
-----------------------------------------------------
Hello Marcus,

I have exactly the same problem.
How were you able to solve it?

Thanks in advance
Greetings Matze

ningpj, 2025-02-05:
dont include ``optional/mmu_menu.cfg``.  These are menus for the original 12864 screen