moggieuk/Happy-Hare #132: pin MMU_SEL_UART is an alias for MMU_GEAR_UART

dajuky, 2023-12-30:
Hi, 
I don't know what to do with this error. Could I ask you for help? 
Thank you very much.

MMU Hardware Initialization -------------------------------
Extruder max_extrude_ratio=20.787584
mcu 'mcu': Starting serial connect
webhooks client 548418194256: New connection
webhooks client 548418194256: Client info {'program': 'Moonraker', 'version': 'v0.8.0-241-gca59516'}
Loaded MCU 'mcu' 124 commands (v0.12.0-57-g9847b449 / gcc: (15:12.2.rel1-1) 12.2.1 20221205 binutils: (2.40-2+18+b1) 2.40)
MCU 'mcu' config: ADC_MAX=4095 BUS_PINS_i2c1=PB6,PB7 BUS_PINS_i2c1a=PB8,PB9 BUS_PINS_i2c2=PB10,PB11 BUS_PINS_i2c3=PA8,PC9 BUS_PINS_sdio=PC12,PD2,PC8,PC9,PC10,PC11 BUS_PINS_spi1=PA6,PA7,PA5 BUS_PINS_spi1a=PB4,PB5,PB3 BUS_PINS_spi2=PB14,PB15,PB13 BUS_PINS_spi2a=PC2,PC3,PB10 BUS_PINS_spi3=PB4,PB5,PB3 BUS_PINS_spi3a=PC11,PC12,PC10 BUS_PINS_spi4=PE13,PE14,PE12 CLOCK_FREQ=180000000 MCU=stm32f446xx PWM_MAX=255 RESERVE_PINS_USB=PA11,PA12 RESERVE_PINS_crystal=PH0,PH1 STATS_SUMSQ_BASE=256 STEPPER_BOTH_EDGE=1
mcu 'mmu': Starting serial connect
Loaded MCU 'mmu' 112 commands (v0.12.0-58-g25bc649c / gcc: (15:12.2.rel1-1) 12.2.1 20221205 binutils: (2.40-2+18+b1) 2.40)
MCU 'mmu' config: ADC_MAX=4095 BUS_PINS_i2c0a=gpio0,gpio1 BUS_PINS_i2c0b=gpio4,gpio5 BUS_PINS_i2c0c=gpio8,gpio9 BUS_PINS_i2c0d=gpio12,gpio13 BUS_PINS_i2c0e=gpio16,gpio17 BUS_PINS_i2c0f=gpio20,gpio21 BUS_PINS_i2c0g=gpio24,gpio25 BUS_PINS_i2c0h=gpio28,gpio29 BUS_PINS_i2c1a=gpio2,gpio3 BUS_PINS_i2c1b=gpio6,gpio7 BUS_PINS_i2c1c=gpio10,gpio11 BUS_PINS_i2c1d=gpio14,gpio15 BUS_PINS_i2c1e=gpio18,gpio19 BUS_PINS_i2c1f=gpio22,gpio23 BUS_PINS_i2c1g=gpio26,gpio27 BUS_PINS_spi0a=gpio0,gpio3,gpio2 BUS_PINS_spi0b=gpio4,gpio7,gpio6 BUS_PINS_spi0c=gpio16,gpio19,gpio18 BUS_PINS_spi0d=gpio20,gpio23,gpio22 BUS_PINS_spi1a=gpio8,gpio11,gpio10 BUS_PINS_spi1b=gpio12,gpio15,gpio14 BUS_PINS_spi1c=gpio24,gpio27,gpio26 CLOCK_FREQ=12000000 MCU=rp2040 PWM_MAX=255 STATS_SUMSQ_BASE=256 STEPPER_BOTH_EDGE=1
mcu 'can0': Starting CAN connect
Created a socket
Loaded MCU 'can0' 114 commands (v0.12.0-57-g9847b449 / gcc: (15:12.2.rel1-1) 12.2.1 20221205 binutils: (2.40-2+18+b1) 2.40)
MCU 'can0' config: ADC_MAX=4095 BUS_PINS_i2c1=PB6,PB7 BUS_PINS_i2c1_PB6_PB7=PB6,PB7 BUS_PINS_i2c1_PB8_PB9=PB8,PB9 BUS_PINS_i2c1_PF1_PF0=PF1,PF0 BUS_PINS_i2c1a=PF1,PF0 BUS_PINS_spi1=PA6,PA7,PA5 BUS_PINS_spi1a=PB4,PB5,PB3 BUS_PINS_spi2=PB14,PB15,PB13 BUS_PINS_spi2a=PC2,PC3,PB10 CANBUS_FREQUENCY=500000 CLOCK_FREQ=48000000 MCU=stm32f072xb PWM_MAX=255 RECEIVE_WINDOW=192 RESERVE_PINS_CAN=PB8,PB9 RESERVE_PINS_crystal=PF0,PF1 STATS_SUMSQ_BASE=256 STEPPER_BOTH_EDGE=1
mcu_temperature 'mcu' nominal base=-280.163934 slope=1342.622951
mcu_temperature 'can0' nominal base=342.711111 slope=-728.000000
Sending MCU 'mcu' printer configuration...
Configured MCU 'mcu' (1024 moves)
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 180, in _connect
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 756, in _connect
    self._send_config(None)
  File "/home/pi/klipper/klippy/mcu.py", line 695, in _send_config
    cmdlist[i] = pin_resolver.update_command(cmd)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/klipper/klippy/pins.py", line 53, in update_command
    return re_pin.sub(pin_fixup, cmd)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/klipper/klippy/pins.py", line 47, in pin_fixup
    raise error("pin %s is an alias for %s" % (
pins.error: pin MMU_SEL_UART is an alias for MMU_GEAR_UART

moggieuk, 2024-01-08:
You have two pins with the same alias name I think.  Normally the UART pins are separate.  The exception is the EASY_BRD shares UART pin so you need to simple remove the MMU_SEL_UART in the aliases file and use the MMU_GEAR_UART on both steppers.  I've update the installer to do this but you will have to edit by hand.

Alternatively you can opt not to use aliases and simply add pin names in `mmu_hardware.cfg`

dajuky, 2024-01-09:
I manually overwrote the pins for my ercf fsb v1.0 board in mmu.cfg and it works, thank you for the answer.

closed 2024-01-09 by dajuky