moggieuk/Happy-Hare #133: Internal error on command:"MMU_HOME"

marcus360, 2023-12-31:
Internal error on command:"MMU_HOME"
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

endstop is working but motors dont't , on easy brd board  
can't home mmu \
[klippy (25).log](https://github.com/moggieuk/Happy-Hare/files/13799962/klippy.25.log)

eisthiele, 2023-12-31:
Same Problem with the BTT MMB.

marcus360, 2023-12-31:
File "/home/pi/klipper/klippy/toolhead.py", line 294, in _advance_flush_time
    clear_history_time = self.clear_history_time
AttributeError: 'MmuToolHead' object has no attribute 'clear_history_time'

its about clearing history time in toolheads, this problem is inside python toolhead.py and mmu_toolhead.py but i am bad ad programing there so just going around problem :/

in line 294 in toolhead.py there is command to ""Free trapq entries that are no longer needed""- 

# Free trapq entries that are no longer needed
        clear_history_time = self.clear_history_time
        if not self.can_pause:
            clear_history_time = flush_time - MOVE_HISTORY_EXPIRE
        free_time = sg_flush_time - self.kin_flush_delay
        self.trapq_finalize_moves(self.trapq, free_time, clear_history_time)
        self.extruder.update_move_time(free_time, clear_history_time)

but in mmu_toolhead.py flush tracking in line 70  -- 
 

# Flush tracking
        self.flush_timer = self.reactor.register_timer(self._flush_handler)
        self.do_kick_flush_timer = True
        self.last_flush_time = self.last_sg_flush_time = 0.
        self.need_flush_time = self.step_gen_time  = 0.

you need add to mmu_toolhead.py self.clear_history_time like this 

# Flush tracking
        self.flush_timer = self.reactor.register_timer(self._flush_handler)
        self.do_kick_flush_timer = True
        self.last_flush_time = self.last_sg_flush_time = 0.
        self.need_flush_time = self.step_gen_time  =  self.clear_history_time = 0. 

it worked for me :)

rwijnhov, 2023-12-31:
Same problem for me also mmb board

rwijnhov, 2023-12-31:
Latest update of happy hare fixes this. Force update

eisthiele, 2023-12-31:
I have the latest Version 2.3.1

marcus360, 2023-12-31:
i had it too

moggieuk, 2024-01-08:
Just catching up with Issue backlog.  Is this still a problem.  It looks like something that was fixed a while ago..?

eisthiele, 2024-01-08:
It work with the actual Version 👍

closed 2024-12-13 by github-actions[bot]