moggieuk/Happy-Hare #135: Feature request - Add macro command to temporary disable/enable Clog detection

cicinovec, 2024-01-02:
Issue: I use MMU with endless spool on. This means the clog detection must be turned on. In general it works great, But there is always a false positive with priming line and in general with first layer.

Solution: Add command to temporary disable clog detection, that could be add to START_PRINT macro (before prime line) and command to enable it at layer 2

DarkSoulRL, 2024-01-04:
Same problem

moggieuk, 2024-01-09:
The "false trigger" isn't actually a false trigger but rather an indication that you are extruding faster than your extruder can handle. That mismatch of desired move and detected move is what triggers the clog.  If you slow that purge down all will be good.

But you can disable clog detection with `MMU_TEST_CONFIG enable_clog_detection=0` and obviously turn it back on again with `=1`

But really I'd start with setting appropriate extruder speeds..

boer0327, 2024-06-05:
this command is very useful for me. it is because I am following Bambu p1s settings to flush filaments. it will always slow-fast-slow-fast for a few times. this will cause clog. when I use MMU_TEST_CONFIG enable_clog_detection=0 when flush start. and use MMU_TEST_CONFIG enable_clog_detection=1 when flush finished. everything is working now. thx moggieuk.

moggieuk, 2026-10-21:
Thanks, and sorry for not closing this sooner. Happy Hare v4 (v4.1 is the current release) has a dedicated command for this, so I'm closing it.

In v4, clog (and tangle) detection is called FlowGuard. You can switch it off and on at any time, for example in your print start macro:

```
MMU_FLOWGUARD ENABLE=0     ; before the prime line / flush
...
MMU_FLOWGUARD ENABLE=1     ; once you're past it
```

Add `UNIT=ALL` if you have more than one MMU unit. Running `MMU_FLOWGUARD` on its own reports the current status. The default comes from `flowguard_enabled` in `mmu_parameters.cfg`.

Note that the v3 commands mentioned earlier in this thread (`MMU_TEST_CONFIG enable_clog_detection=...`, `MMU_ENCODER ENABLE=...`) don't exist in v4. Also, a trigger during fast purges is usually a real sign that the extruder can't keep up, so slowing the purge is still worth trying.

If there's something v4 still doesn't do for you here, please open a new feature request against v4, which is being actively worked on.

closed 2026-10-21 by moggieuk