moggieuk/Happy-Hare #140: Failed to detect filament run-out

LostFool, 2024-01-06:
Was running a print last night with low filament. When it ran-out, the print just kept going. Got some failures to swap filament as well and the print just kept going even when the swap failed. Regardless, filament run-out should have been detected.

I feel like the longer a printer has been on, regardless of soft restart, the more likely Happy Hare is to start failing to operate correctly. I haven't restarted this printer in 4+ days, but did make some config changes before this last print, and then these errors started. Mind you I have done multiple prints in the time alive. (note by soft restart I made config changes and did the save-restart but didn't hard restart by hitting the power switch).

[logs and config.zip](https://github.com/moggieuk/Happy-Hare/files/13850313/logs.and.config.zip)

moggieuk, 2024-01-09:
Thanks for suppling all the logs and config.  Can you point me to the timestamp of where you think is started to go wrong. Also did you do any `MMU_TEST_CONFIG` changes after reboot or can I assume that mmu_parameters is the config you were printing with?

LostFool, 2024-01-13:
I do not do any MMU_Test_Config during this run. 
I have tried it in runs after this when I turned on sync during print, TPU is too stretchy, so I was testing turning on sync and found I do need it. I had some crashes with it turned on (and a few other features) so I'm working back into optimal settings.

As I understand the issues started early in the mmu_log file. I am not sure if it started any earlier, but the actual print was started around 9pm the day before this log. Line 35 is the first error, and I assume that is the actual start of the problems.
The equivalent in the klippy log appears to be line 12370. Though without time stamps this one is harder to confirm.

It definitely should have read 0 at a certain point, as it ran out of filament. I would expect this to pause the print.

LostFool, 2024-01-31:
So I haven't done any printing for awhile due to extreme cold weather in the garage. Printer has been off.

I went out yesterday, turned on the printer, loaded up fresh dry filament, ran loading and unloading tests then let the printer sit idle overnight. Today I try a print and extruder syncing isn't working, some failures to unload.
I started to see the typical issues I see when the printer has been on for a long period of time. I'm not sure what causes these issues. Stale cache, memory?

Hard restart solves the problems for now.

moggieuk, 2024-02-06:
Sorry for the delay.  Firstly on your runout issue.  I've done a lot of work lately on restructuring the way runout works so Id definitely recommend your force an update and try again.  Because of the restructure it's hard to see what might have been happening.

On the printer overnight issue... I've really no idea.  I often leave my printer on for days without using and then its ready to go the next time.  Is it possible your SD card is full?    Restarting the printer (or rather the rpi) would clear our some space.

LostFool, 2024-02-06:
No worries! All in your own time. Thank you for your work.
I saw the update last night but wasn't ready to embrace it due to time
constraints.
I'm thinking about wiping my rpi and starting with a fresh system. I think
it's due in since it's been a learning system the whole way.
But I didn't think to check space... I'll look into it and let you know.

LostFool, 2024-02-24:
A lot of problems seem to have gone away. I have had the printer on for a number of days, changed configs and rarely restart the printer. No issues yet.
I'm betting on Happy Hare firmware updates solved issues.

closed 2024-03-02 by LostFool