moggieuk/Happy-Hare #141: Unnecessary waiting for the hotend to cool down.

DarkSoulRL, 2024-01-08:
ABS was printed at a temperature of 280 degrees.
At the end of the work we see this.
[image]

The current version of HH began to wait for the nozzle to cool down to 200g after printing in order to extract the filament. Why is this done, it is not required. The same thing happens when printing starts, if the nozzle was heated above 200 degrees, HH waits for it to cool down to 200 degrees.

moggieuk, 2024-01-09:
So this is while printing?   Can you attach the `mmu.log`.   The logic around what temperature to be at is surprisingly complex...  the info you have giving me gives me a hint but the log would be better..

DarkSoulRL, 2024-01-09:
look at 23:11:44, 
current_temp=270.76
Warning:  Automatically heating extruder to minimum temp (200.0)
[mmu.zip](https://github.com/moggieuk/Happy-Hare/files/13869473/mmu.zip)

bartoszwas, 2024-01-12:
I had same error, it was PRINT_END macro issue when I turned off heaters first then try to unload. When i switched commands to first unload and then turn off heaters everytihng is fine. Just target temp was set to 0 so it had to reheat again.

DarkSoulRL, 2024-01-12:
not the case
[image]

bartoszwas, 2024-01-12:
in your log there is:

18:49:10 - - TRACE: _ensure_safe_extruder_temperature: current_temp=268.71, paused_extruder_temp=None, **current_target_temp=0.0**, klipper_minimum_temp=170.0, default_extruder_temp=200.0, source=auto
18:49:10 Warning: Automatically heating extruder to minimum temp (200.0)

So something sets temp to 0.0 before unload. Don't you have doubled PRINT_END macro? Maybe Slicer turns off the heaters?

closed 2024-01-22 by DarkSoulRL