moggieuk/Happy-Hare #144: ThumperBlock thicker version

Anollya, 2024-01-08:


moggieuk, 2024-01-09:
Actually, I'm going a different way.  Now there is a 23mm version of TB you can do away with the "h" suffix.  TB is now plug and play with TD (although you might want to optimize `gate_parking_distance` for the two.

moggieuk, 2026-10-21:
Thanks, and sorry for leaving this open. The Happy Hare v4 installer (v4.1 is the current release) covers this, so I'm closing it.

When you pick ERCF v2 as the MMU type in menuconfig, there's a "Project Options" menu with a "Thumper Blocks" option. Selecting it sets the default `gate_parking_distance` and servo angles for Thumper Blocks. As I mentioned above, the current 23mm Thumper Blocks are plug-and-play, so you don't need a separate "thicker" option. You may still want to fine-tune `gate_parking_distance` for your build.

If there's anything else the v4 installer should ask about for your setup, please open a new feature request against v4, which is being actively worked on.

closed 2026-10-21 by moggieuk