moggieuk/Happy-Hare #145: Internal error on command:"MMU_CHANGE_TOOL" when toolhead_sensor_to_nozzle is reduced

jfwells, 2024-01-09:
Trying to reduce blobs in purge tower had me reducing toolhead_sensor_to_nozzle slightly (I have 1 filament sensor after the extruder gear -- the model originally posted on the filametrix page). at 51 it works. At 50 I consistently get a klipper shutdown on completion of the first toolchange.

Klippy logs attached (ignore the EBBCan errors, they come after klipper cuts power)

[klippy (10).log](https://github.com/moggieuk/Happy-Hare/files/13880672/klippy.10.log)
[mmu_parameters.txt](https://github.com/moggieuk/Happy-Hare/files/13880688/mmu_parameters.txt)

jfwells, 2024-01-20:
Do you need any further debug info on this? It's reproducible for me, and raising back up to 51, the internal error goes away. I assume something is unexpectedly negative or divided by zero somewhere.

dominikacri, 2024-02-24:
Same problem here. 
Internal error on command:"T0"
Internal Error on WebRequest: gcode/script
Traceback (most recent call last):
  File "/home/dominik/klipper/klippy/extras/mmu.py", line 4614, in _wrap_sync_extruder_to_gear
    yield self
  File "/home/dominik/klipper/klippy/extras/mmu.py", line 4379, in _trace_filament_move
    trig_pos = hmove.homing_move(pos, speed, probe_pos=True, triggered=homing_move > 0, check_triggered=True)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dominik/klipper/klippy/extras/homing.py", line 94, in homing_move
    self.toolhead.drip_move(movepos, speed, all_endstop_trigger)
  File "/home/dominik/klipper/klippy/toolhead.py", line 529, in drip_move
    self.lookahead.flush()
  File "/home/dominik/klipper/klippy/toolhead.py", line 176, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/dominik/klipper/klippy/toolhead.py", line 351, in _process_moves
    self._update_drip_move_time(next_move_time)
  File "/home/dominik/klipper/klippy/toolhead.py", line 508, in _update_drip_move_time
    self._advance_move_time(npt)
  File "/home/dominik/klipper/klippy/toolhead.py", line 312, in _advance_move_time
    self._advance_flush_time(flush_time)
  File "/home/dominik/klipper/klippy/toolhead.py", line 303, in _advance_flush_time
    m.flush_moves(flush_time, clear_history_time)
  File "/home/dominik/klipper/klippy/mcu.py", line 938, in flush_moves
    raise error("Internal error in MCU '%s' stepcompress"
mcu.error: Internal error in MCU 'mcu' stepcompress

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dominik/klipper/klippy/extras/mmu.py", line 3807, in _load_sequence
    self._load_extruder()
  File "/home/dominik/klipper/klippy/extras/mmu.py", line 3625, in _load_extruder
    _,fhomed,measured,_ = self._trace_filament_move("Homing to toolhead sensor", self.toolhead_homing_max, motor=motor, homing_move=1, endstop_name=self.ENDSTOP_TOOLHEAD)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dominik/klipper/klippy/extras/mmu.py", line 4365, in _trace_filament_move
    with self._wrap_sync_extruder_to_gear(motor in ["gear+extruder", "extruder"], extruder_only=(motor == "extruder")):
  File "/usr/lib/python3.11/contextlib.py", line 155, in __exit__
    self.gen.throw(typ, value, traceback)
  File "/home/dominik/klipper/klippy/extras/mmu.py", line 4617, in _wrap_sync_extruder_to_gear
    self.mmu_toolhead.sync_extruder_to_gear(None)
  File "/home/dominik/klipper/klippy/extras/mmu_toolhead.py", line 277, in sync_extruder_to_gear
    self.flush_step_generation()
  File "/home/dominik/klipper/klippy/toolhead.py", line 363, in flush_step_generation
    self._flush_lookahead()
  File "/home/dominik/klipper/klippy/toolhead.py", line 357, in _flush_lookahead
    self.lookahead.flush()
  File "/home/dominik/klipper/klippy/toolhead.py", line 176, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/dominik/klipper/klippy/toolhead.py", line 351, in _process_moves
    self._update_drip_move_time(next_move_time)
  File "/home/dominik/klipper/klippy/toolhead.py", line 508, in _update_drip_move_time
    self._advance_move_time(npt)
  File "/home/dominik/klipper/klippy/toolhead.py", line 312, in _advance_move_time
    self._advance_flush_time(flush_time)
  File "/home/dominik/klipper/klippy/toolhead.py", line 303, in _advance_flush_time
    m.flush_moves(flush_time, clear_history_time)
  File "/home/dominik/klipper/klippy/mcu.py", line 938, in flush_moves
    raise error("Internal error in MCU '%s' stepcompress"
mcu.error: Internal error in MCU 'mcu' stepcompress

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dominik/klipper/klippy/webhooks.py", line 256, in _process_request
    func(web_request)
  File "/home/dominik/klipper/klippy/webhooks.py", line 436, in _handle_script
    self.gcode.run_script(web_request.get_str('script'))
  File "/home/dominik/klipper/klippy/gcode.py", line 229, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
[... 18 lines cut ...]
  File "/home/dominik/klipper/klippy/extras/mmu.py", line 4730, in _select_and_load_tool
    self._load_sequence()
  File "/home/dominik/klipper/klippy/extras/mmu.py", line 3824, in _load_sequence
    self._servo_up()
  File "/home/dominik/klipper/klippy/extras/mmu.py", line 1647, in _servo_up
    self._movequeues_wait_moves()
  File "/home/dominik/klipper/klippy/extras/mmu.py", line 1134, in _movequeues_wait_moves
    self.mmu_toolhead.wait_moves()
  File "/home/dominik/klipper/klippy/toolhead.py", line 480, in wait_moves
    self._flush_lookahead()
  File "/home/dominik/klipper/klippy/toolhead.py", line 357, in _flush_lookahead
    self.lookahead.flush()
  File "/home/dominik/klipper/klippy/toolhead.py", line 176, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/dominik/klipper/klippy/toolhead.py", line 351, in _process_moves
    self._update_drip_move_time(next_move_time)
  File "/home/dominik/klipper/klippy/toolhead.py", line 508, in _update_drip_move_time
    self._advance_move_time(npt)
  File "/home/dominik/klipper/klippy/toolhead.py", line 312, in _advance_move_time
    self._advance_flush_time(flush_time)
  File "/home/dominik/klipper/klippy/toolhead.py", line 303, in _advance_flush_time
    m.flush_moves(flush_time, clear_history_time)
  File "/home/dominik/klipper/klippy/mcu.py", line 938, in flush_moves
    raise error("Internal error in MCU '%s' stepcompress"
mcu.error: Internal error in MCU 'mcu' stepcompress

closed 2026-10-21 by moggieuk