moggieuk/Happy-Hare #146: [Not an issue] Endless spool on load not working when gate status is unknown

bartoszwas, 2024-01-11:
Hello,

First of all thanks to moggieuk for tremendous amount of work with this software <3

So to point. As I use only level 2 of persistence so my gates always have unknown status on power on. I think that causes endless spool to not working properly when gate is empty. My build is for production at my worksite it has 4 gates and all of them in one endless spool group.

So is it possible to make it check other gate when it's status is unknown but not empty?

Thanks in advance, and keep it going.

moggieuk, 2024-01-13:
Thanks.  The endless spool on load is new and perhaps not completely defect-free yet, but to your point...

If you have a persistence level of 2 (BTW I run at 4 and it is so nice just to pick things up from where you left them) then indeed gates will be in an unknown state on startup.

The simple solution is to do a `MMU_CHECK_GATE` on startup, perhaps delayed gcode or even put in print start.  This will check and set the state of each gate...

I may be able to get HH to do a "MMU_CHECK_GATE GATE=x" behind the scenes when the gate is unknown state if endlessSpool is enabled. This could harmless be put before the current check.  I'll add to the todo list but can't guarantee when I'll be able to get this shipped...

bartoszwas, 2024-01-13:
Thanks. I think check gate is the best solution. As this machine has only 4 gates it won’t take that much time to check them.

according to level 2 of persistence I thought it will be good. It picked up filament from gate 0 on start.  The reason I use level 2 is there are more people to operate this machine. If filament ends they load new spool but often forget to set it to available so machine thinks it’s empty and doesn’t even try to pick it up…

moggieuk, 2024-01-18:
Ok, persistence level of 2 makes sense for you.  Yes, I think running a complete `MMU_CHECK_GATE` on startup is the quickest and best solution.   Many folks do this in print start (where you can also only check the tools that are actually used in the print ... see `!referenced_tools!` feature) but you can also do with a delayed gcode - perhaps running a few seconds after startup.

moggieuk, 2024-02-05:
Closing issue

closed 2024-02-05 by moggieuk