moggieuk/Happy-Hare #147: Hi,GUYS，I create a new Electrical rewind

muzixiaoyang, 2024-01-11:
[https://github.com/muzixiaoyang/3Dprinter-Electric-rewind---filament-rewind---multi-color-printing](url)
[https://www.bilibili.com/video/BV1MT4y1n7Rv/?spm_id_from=333.999.0.0](url)

moggieuk, 2024-01-13:
Wow. That is very nice!

I watched the video and it looks very clever.  Have you shared on the Voron Discord?

muzixiaoyang, 2024-01-13:
Thank you sir, I haven't shared VORON Discord yet. I am preparing to translate it and share it on Discord. At present, I have prepared your Type-B hardware

moggieuk, 2024-01-18:
Great. HH support for "Type-B" solutions is in alpha.  The main code I need time to test is the drive stepper switching logic..

muzixiaoyang, 2024-01-28:
Thank you, sir, I have prepared two TYPE-B schemes and created two new PCB boards, ams and ams lite, maybe we can communicate further :D

[image]
[image]ding FU.JPG…]()
[image]

thiagolocatelli, 2024-01-31:
WOW! This looks nice! Reversed engineered the AMS feeder. Clever.

moggieuk, 2024-02-06:
> Thank you, sir, I have prepared two TYPE-B schemes and created two new PCB boards, ams and ams lite, maybe we can communicate further :D

Happy to keep in touch because I can see this being a popular edition and I'd love to be able to drive it with Happy Hare...

muzixiaoyang, 2024-02-07:
Yes sir, I am sorting out the repository of AMS-LITE

thiagolocatelli, 2024-02-07:
@muzixiaoyang do you have new videos?

muzixiaoyang, 2024-02-07:
Yes, I have the latest introductory video, but in Chinese, currently preparing to shoot a video in English, sir
https://www.bilibili.com/video/BV1Mv421i7iz

thiagolocatelli, 2024-02-07:
@muzixiaoyang that is not very far from what I had working! You have made way more progress. Also, Mellow has a new board that you can use on your project too. I have been beta testing it.

[Video Loading filament](https://cdn.discordapp.com/attachments/1155906995950465146/1155922703040122960/IMG_8214.mov?ex=65d39368&is=65c11e68&hm=b3dd59016265f412f49ffea3a5e9037feaf66fde1a57b9b618b515a701d21705&)

[Video Homing filament to toolhead](https://cdn.discordapp.com/attachments/1155906995950465146/1160952392766988391/IMG_8412.mov?ex=65d36aab&is=65c0f5ab&hm=85e357947d675d02429ffb5cfcb1e78f6a430d3c4f040e04205979982fbf3ec1&)

[Video Buffer/4-in-1-out internals](https://cdn.discordapp.com/attachments/1155906995950465146/1160021782166765619/IMG_8389.mov?ex=65d007f9&is=65bd92f9&hm=bc3970a1b5a8114891fe5ee84c0a632f7c136f36773b4c657f0111be55af27da&)

[image]

[image]

[image]

muzixiaoyang, 2024-02-08:
Great,sir,i make a new 4-in-1-out,maybe I can share it with you :D
[image]

muzixiaoyang, 2024-02-08:
Hi sir, can I add your discord?

thiagolocatelli, 2024-02-08:
My discord is ".doublet" or just search for DoubleT

muzixiaoyang, 2024-02-08:
Okay, I added
My discord is "muzixiaoyang"
Looking forward to communicating with you

closed 2026-10-21 by moggieuk