moggieuk/Happy-Hare #148: `History` from moonraker doesn't work when error ocours while using MMU

jmceara, 2024-01-11:
If I start a print job using HH and for some reason print is paused (ex: clog or load/unload error), after solveing the issue and resume print job, everything works and print job is done until the end (print_end macro is ran fine).
But when this happens (any error on MMU), the job is not accounted in History plugin of moonraker. Not as 'error', not as 'success', just doesn't register at all.
If the print job ran without any error, the job is registered fine in `History` componente of moonraker.
Already opened a ticket in moonraker/mainsail discord, they said that is probably a HH error/bug and not in moonraker.
[Logs.zip](https://github.com/moggieuk/Happy-Hare/files/13907873/Logs.zip)

moggieuk, 2024-01-13:
Thanks.  I'm not very familiar with moonraker history interface and whether there is any way to "influence" it.  It would certainly be nice to notify it of an MMU problem but the way the MMU works, is that if a problem occurs, the print is simply paused.  After fixing the MMU problem the job is resumed.  This will look quite normal to moonraker.

I think what would be needed would be for moonraker to listen to pause events and to record those against a job so that it would be, for example, "successful with 5 pauses"

See what the moonraker folks think about this...

moggieuk, 2024-02-05:
I'm curious if you have any update on this.  How are "events" sent to moonraker history?

moggieuk, 2026-10-21:
Thanks for reporting this, and sorry it went quiet. I'm closing it because it was reported against a much older version, and I can't see a way for current Happy Hare to cause it.

Here's why. Moonraker builds its job history from Klipper's `print_stats` state (printing, paused, complete, error). In Happy Hare v4 (v4.1 is the current release), an MMU error simply pauses the print with Klipper's normal `PAUSE`, and resuming continues the same job. Happy Hare never resets or rewrites `print_stats` and doesn't talk to the history component. A print that paused for an MMU problem and then finished should be recorded as a normal completed job.

If you still see jobs missing from history on v4, please open a new bug report. Attach `mmu.log`, `klippy.log` and `moonraker.log` from that print. Things like a Klipper firmware restart during recovery are the usual suspects. Recording the number of pauses against a job would need to be a Moonraker feature, so that one is best raised with the Moonraker project.

If there's something you'd like Happy Hare v4 to do here, please open a new feature request against v4, which is being actively worked on.

closed 2026-10-21 by moggieuk