moggieuk/Happy-Hare #149: _MMU_PRE_UNLOAD and _MMU_POST_LOAD macros not running on toolchange

121Jigawatt, 2024-01-12:
When changing tools with the provided default MMU_CHANGE_TOOL for T0 the macros that I have located in the _MMU_PRE_UNLOAD and _MMU_POST_LOAD don't run.

When I manually run the pre and post macros it runs correctly but not during a tool change.

moggieuk, 2024-01-17:
Current logic was only to run them while printing I think.  That was an oversight.  I'm pushing an update today to address that...

121Jigawatt, 2024-01-17:
Thank you, I have it set for happy hair to do the tip forming for the filament cutting but have some macros in there to use a purge system I'm making

moggieuk, 2024-01-18:
Ok. I think with v2.4.1 release we should all be good now.  Let me know.  If I don't hear back in few days I'm going to close this issue.

121Jigawatt, 2024-01-19:
After updating to 2.4.1 I'm now getting this error.
`MMU Hardware Initialization -------------------------------
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/server/klipper/klippy/klippy.py", line 175, in _connect
    self._read_config()
  File "/home/server/klipper/klippy/klippy.py", line 141, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/server/klipper/klippy/klippy.py", line 130, in load_object
    self.objects[section] = init_func(config.getsection(section))
  File "/home/server/klipper/klippy/extras/mmu.py", line 6093, in load_config
    return Mmu(config)
  File "/home/server/klipper/klippy/extras/mmu.py", line 652, in __init__
    self._setup_mmu_hardware(config)
  File "/home/server/klipper/klippy/extras/mmu.py", line 665, in _setup_mmu_hardware
    self.mmu_toolhead = MmuToolHead(config, self.homing_extruder)
  File "/home/server/klipper/klippy/extras/mmu_toolhead.py", line 34, in __init__
    self.move_queue = toolhead.MoveQueue(self) # Happy Hare: Use base class MoveQueue
AttributeError: module 'toolhead' has no attribute 'MoveQueue'
webhooks client 140041665687856: New connection
webhooks client 140041665687856: Client info {'program': 'Moonraker', 'version': 'v0.8.0-267-g66de18f'}`
I have re-installed klipper in troubleshooting so it is a fresh install. Is this something in klipper or HappyHare?

121Jigawatt, 2024-01-23:
All good now, probably not very many people have this issue but make sure klipper and happy-hare is on the latest version.
There was a name change of a toolhead param in klipper that broke happy hare but it has since been fixed.

closed 2024-01-23 by 121Jigawatt