moggieuk/Happy-Hare #152: Possible bug in GATE to LED mapping (typical programmers "1")

evzenmayer, 2024-01-16:
Hi,

I'm suspicious that there's a minor bug in GATE to LED mapping when LED are in reverse order. The LED 1 (= gate 7 in my setup) never lights up on entry - no matter if status or filament color is expected to be there. The LED work OK when lighted by hand by GCODE. The rest of LEDs (2-8 = gates 0-6) works as expected.

My config:

[neopixel mmu_leds]
pin: mmu:MMU_NEOPIXEL
chain_count: 8			# Number gates x1 or x2 + 1 (if you want status)
color_order: GRB		# Set based on your particular neopixel specification

[mmu_leds]
num_gates: 8
led_strip: neopixel:mmu_leds
entry_range: 8-1
frame_rate: 24

Many thanks for looking to that.

Evzen

evzenmayer, 2024-01-16:
Only appears when not all gates are initialized => "repaired" by itself in my case ;-)

moggieuk, 2024-01-17:
Could be a bug... I'll take a look. Thanks.

moggieuk, 2024-02-05:
Bug fixed so closing issue

closed 2024-02-05 by moggieuk