moggieuk/Happy-Hare #153: pre gate sensor with just one pin, possible?

bartoszwas, 2024-01-16:
Hey,

Is it possible to use just one pin for multiple pre gate sensors connected in series? As MMU knows which gate is already in use in theory it should be possible to do that. Asking because my Easy Board has just one spare pin to use and I would like to not change it to BTT if it's not necessary.

o1hitman1o, 2024-01-16:
I would say no. The pre gate is used to determine if filament is loaded in the gate ready to use. If you have filament in gate 1 but are using gate 2  with no filament  loaded it would think there is filament loaded in gate 2 due to gate one...  You don't have to use the pre gates.

bartoszwas, 2024-01-16:
Yeah... I thought in case of filament runout...

closed 2024-01-16 by bartoszwas