moggieuk/Happy-Hare #154: Unable to reverse LED order for gates

Ekodas, 2024-01-17:
I am unable to reverse the order of the LED's for my gates.  

The LED Support page states:  Typically the first LED would be for gate 0 but the order can be reversed by setting reverse_gate_order:1 in mmu_software.cfg.  This does not work and induces an error on firmware startup.

In the mmu_software.cfg there are 2 lines that deal with reversing order:
variable_exit_reverse_order: 0
variable_entry_reverse_order: 0

I have tried changing each of these values (individually and together) to 1 however it had no effect on the LED order.  

I'm running an ERCF v2 with the BTT MMB.  The LED's are working properly.  Please let me know if I'm missing something or doing something wrong.  Also let me know if you want a copy of my cfg files for troubleshooting.

Thanks in advance!

moggieuk, 2024-01-17:
The last update made this extremely easy.  I'm finishing off a new doc page to explain but you need only change this section at the bottom of `mmu_hardware.cfg` now:
```
[mmu_leds]
num_gates: 8
led_strip: neopixel:mmu_leds
exit_range: 1-8
#entry_range:
status_index: 9
frame_rate: 24
```
If I want to reverse the order of the "exit leds" change line to:
`exit_range: 8-1`

Easy!

Ekodas, 2024-01-17:
Knew it had to be something simple.  Thanks for the quick response!  Great job on the project as a whole!

moggieuk, 2024-01-17:
Thanks.  Actually just so you know, I dived deeper and my comment wasn't the complete solution. There was a bug (or rather a maths error).  I just pushed v2.4.1 with a couple of updates and this was one.

moggieuk, 2024-02-05:
Bug fixed so closing issue

closed 2024-02-05 by moggieuk

TravisWilder, 2024-02-08:
> The last update made this extremely easy. I'm finishing off a new doc page to explain but you need only change this section at the bottom of `mmu_hardware.cfg` now:
>
> ```
> [mmu_leds]
> num_gates: 8 [...]

So when I have entry and exit LEDs I can do like 
> exit_range: 1-8
> entry_range: 10-17
> status_index: 9

or 
> exit_range: 9-16
> entry_range: 1-8
> status_index: 17