moggieuk/Happy-Hare #155: Sync to extruder not enabled at print start

lettore, 2024-01-17:
I just updated to latest version, and found that Sync is not enabled at print start but later.
I use a print_start macro with integrated purge and Sync is not running during this macro as before the update.
I should add something in print_start macro to enable sync mode?

moggieuk, 2024-01-18:
I don't think this logic has change but perhaps you particular start-up is very slightly different.  Here's what's happening and what you can do:
1. The automatic detection of print start is on an "event" handler and I cannot safely do certain things at that time.  That includes movement the servo.  Therefore, I do what I can and put the sync/servo move into the move queue and it happens a  couple of gcode instructions later.
2. You can force this simply by (leaving auto-detect on) and adding a `_MMU_PRINT_START` at the top of your print_start macro.  This will sync and servo right at the very beginning. This essentially jumps the "started" state and goes straight to "printing"   The automatic call that will happen later will do nothing and is harmless.

Can you let me know if this works ok. 

(I will look again at dong the sync/servo automatically but I'm sure when I tried I bombed klipper...)

lettore, 2024-01-18:
Yes adding `_MMU_PRINT_START` in the `PRINT_START`  macro fix the issue.