moggieuk/Happy-Hare #158: Error on first change tool AttributeError: 'NoneType' object has no attribute 'test'

deathfrozen, 2024-01-19:
For some time now, I have been consistently getting the Printer is shut down error at the beginning of printing.
I noticed patterns: during printing, changing the tool does not cause an error.
Tool replacement does not cause an error during downtime.
The error occurs only at the very beginning of printing when the selector is moved in cases such as MMU_CHECK_GATE or MMU_SELECT or MMU_CHANGE_TOOL.
[klippy (16).log](https://github.com/moggieuk/Happy-Hare/files/13994308/klippy.16.log)

deathfrozen, 2024-01-21:
I added CODE G4 P1000 to the beginning of the START_PRINT macro and it fixed all the problems.

moggieuk, 2024-01-21:
Thanks for the logs.  Does seem to be deep down in klipper ??? but I'll take a look.  What version of Klipper are you running?

deathfrozen, 2024-01-21:
I recently updated and now I have versions -
klipper v0.12.0-85
happy-hare v2.4.1-7
But there were problems on several previous versions too.
Thank you for the wonderful software.

moggieuk, 2024-02-05:
I have heard of a similar problem that was solved by starting with a `mmu_home` operation.  There is something deep down in klipper that is not initialized yet.  Waiting for the MMU_HOME seems to fixed it.  I'm leaving issue open because I want to get to the root cause.

closed 2024-11-29 by github-actions[bot]