moggieuk/Happy-Hare #159: Servo is not moved to 'up' position when doing a filametrix toolchange

jwhite, 2024-01-20:
Reproduction Steps:

1) Reboot printer
2) Raise nozzle temperature
3) Observe that servo position in in 'move' and filament locked
3) Issue a toolchange command (my first macro in START_PRINT)
4) Tip is cut and reverse extrusion occurs

Result : 

Either the ECAS fitting is popped off or something else catastrophically fails as the filament is essentially locked in the servo 'move' position while the reverse extrude after the cut occurs.

A possible solution is to put the servo in the 'up' position during a cut tip toolchange.

moggieuk, 2024-01-21:
That's strange.  The tip forming (inside happy hare) always adjusts servo position prior to the tip forming macro (form or cut).  It depends on the "sync_form_tip" setting.   If sync is enabled servo will be down, it not it should be moved up.

I'll test but it's not clear to me what you are seeing.  The `mmu.log` would be helpful.

jwhite, 2024-01-23:
>sync_form_tip: 0			# Synchronize during standalone tip formation (initial part of unload)

Interesting.  I have been able to repro this a few time now.  I will wait until it happens again and include logs.

jwhite, 2024-01-24:
```
- - TRACE: _bootup_tasks()
(\_/)
( *,*)
(")_(") Happy Hare Ready
(\_/)
( *,*)
(")_(") Happy Hare Ready
TTG Map:
Gates: |#0 |#1 |#2 |#3 |#4 |#5 |#6 |#7 |#8 |
Tools: |T0 |T1 |T2 |T3 |T4 |T5 |T6 |T7 |T8 |
Avail: | S | B | B | ? | B | S | ? | ? | ? |
Selct: ------------------------------------- NOT HOMED
TTG Map:
Gates: |#0 |#1 |#2 |#3 |#4 |#5 |#6 |#7 |#8 |
Tools: |T0 |T1 |T2 |T3 |T4 |T5 |T6 |T7 |T8 |
Avail: | S | B | B | ? | B | S | ? | ? | ? |
Selct: ------------------------------------- NOT HOMED
- DEBUG: Job State: READY -> INITIALIZED (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: ready, Position saved: None, z_hop @0.0mm, pause_resume: False, Idle timeout: 1800.00s)
- - TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='initialized' OLD_STATE='ready'
- DEBUG: Disabled encoder sensor. Status: {'encoder_pos': 0.0, 'detection_length': 11.0, 'min_headroom': 11.0, 'headroom': 16.0, 'desired_headroom': 5.0, 'detection_mode': 2, 'enabled': True, 'flow_rate': 0}
- DEBUG: Setting servo to move (filament hold) position at angle: 95
- - TRACE: Processing idle_timeout 'printing' event
- DEBUG: Requesting the following gate/spool_id pairs from Spoolman: [(0, 20), (1, 15), (2, 18), (3, 2), (4, 12), (5, 19)]
Stats 183538.0: gcodein=0  mcu: mcu_awake=0.007 mcu_task_avg=0.000008 mcu_task_stddev=0.000049 bytes_write=14624 bytes_read=8207 bytes_retransmit=9 bytes_invalid=0 send_seq=522 receive_seq=521 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=180002370 sb2040: mcu_awake=0.017 mcu_task_avg=0.000021 mcu_task_stddev=0.000033 bytes_write=1429 bytes_read=4984 bytes_retransmit=0 bytes_invalid=0 send_seq=149 receive_seq=149 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=12000103 adj=11999619 mmu: mcu_awake=0.012 mcu_task_avg=0.000020 mcu_task_stddev=0.000032 bytes_write=1474 bytes_read=4910 bytes_retransmit=9 bytes_invalid=0 send_seq=153 receive_seq=153 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=10 upcoming_bytes=8 freq=47993232 adj=47994383  toolhead: temp=34.9 raspberry_pi: temp=68.2 mcu_temp: temp=40.2 heater_bed: target=0 temp=28.9 pwm=0.000 sysload=0.24 cputime=13008.836 memavail=1534468 print_time=7.608 buffer_time=0.545 print_stall=0 extruder: target=0 temp=37.9 pwm=0.000
- DEBUG: Received gate map update from Spoolman: {0: {'spool_id': 20, 'material': 'PLA+', 'color': '48FF48'}, 1: {'spool_id': 15, 'material': 'PLA+', 'color': '000000'}, 2: {'spool_id': 18, 'material': 'PLA+', 'color': 'E27304'}, 3: {'spool_id': 2, 'material': 'ABS+', 'color': '000000'}, 4: {'spool_id': 12, 'material': 'ABS+', 'color': '86FF4A'}, 5: {'spool_id': 19, 'material': 'PETG', 'color': 'FFFFFF'}}
- - TRACE: Running macro: _MMU_GATE_MAP_CHANGED GATE=-1
Stats 183539.0: gcodein=0  mcu: mcu_awake=0.007 mcu_task_avg=0.000008 mcu_task_stddev=0.000049 bytes_write=17726 bytes_read=8827 bytes_retransmit=9 bytes_invalid=0 send_seq=595 receive_seq=595 retransmit_seq=2 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=180002189 sb2040: mcu_awake=0.017 mcu_task_avg=0.000021 mcu_task_stddev=0.000033 bytes_write=1435 bytes_read=5080 bytes_retransmit=0 bytes_invalid=0 send_seq=150 receive_seq=150 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=12000066 adj=11999757 mmu: mcu_awake=0.012 mcu_task_avg=0.000020 mcu_task_stddev=0.000032 bytes_write=1508 bytes_read=5076 bytes_retransmit=9 bytes_invalid=0 send_seq=156 receive_seq=156 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=47993061 adj=47992474  toolhead: temp=34.5 raspberry_pi: temp=69.6 mcu_temp: temp=40.4 heater_bed: target=0 temp=28.9 pwm=0.000 sysload=0.30 cputime=13008.931 memavail=1538004 print_time=7.608 buffer_time=0.000 print_stall=0 extruder: target=0 temp=37.8 pwm=0.000
- - TRACE: Processing idle_timeout 'ready' event
- DEBUG: Unloading tool unknown
- - TRACE: Running macro: _MMU_PRE_UNLOAD
Attempting to recover filament position...
Attempting to recover filament position...
- - TRACE: (toolhead sensor detects filament)
MMU [T?] <<< [En] <<<<<<< [Ex] << [Ts] << [Nz] LOADED 0.0mm (e:0.0mm)
MMU [T?] <<< [En] <<<<<<< [Ex] << [Ts] << [Nz] LOADED 0.0mm (e:0.0mm)
Unloading filament...
Unloading filament...
- - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Unloading' OLD_ACTION='Idle'
MMU [T?] <<< [En] <<<<<<< [Ex] << [Ts] << [Nz] LOADED 0.0mm (e:0.0mm)
MMU [T?] <<< [En] <<<<<<< [Ex] << [Ts] << [Nz] LOADED 0.0mm (e:0.0mm)
- - TRACE: (toolhead sensor detects filament)
- - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Forming Tip' OLD_ACTION='Unloading'
- - TRACE: _ensure_safe_extruder_temperature: current_temp=37.84, paused_extruder_temp=None, current_target_temp=0.0, klipper_minimum_temp=180.0, default_extruder_temp=200.0, source=auto
Warning: Automatically heating extruder to default temp (200.0)
- - TRACE: Processing idle_timeout 'printing' event
- - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Heating' OLD_ACTION='Forming Tip'
Waiting for extruder to reach target (default) temperature: 200.0
Waiting for extruder to reach target (default) temperature: 200.0
Stats 183569.0: gcodein=0  mcu: mcu_awake=0.156 mcu_task_avg=0.000140 mcu_task_stddev=0.000104 bytes_write=119604 bytes_read=29511 bytes_retransmit=9 bytes_invalid=0 send_seq=3136 receive_seq=3136 retransmit_seq=2 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=1800016 [... cut ...]
Unload of -855.1mm filament successful (encoder measured 913.6mm)
- - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Idle' OLD_ACTION='Unloading'
- DEBUG: Deactivating spool ...
- - TRACE: Running macro: _MMU_POST_UNLOAD
- - TRACE: Processing idle_timeout 'ready' event
```

As you can see on line 21, the servo position is set to 'move' (HOLDING THE FILAMENT IN PLACE) and then the TIP_CUT macro runs, which tries to lift the tip up to the blade against resistance at the servo, blowing off the ECAS.

Unless you see anything different?

moggieuk, 2024-02-05:
Catching up on Issues.  Thank you for being such an prolific test / issue submitter!! 

I'm reviewing all your submissions and I can't see how this can occur now (at least with the latest main branch).  When the extruder is unloaded in must (now) set the servo in either the up/down position.

I tried to stage this setup based on what I see and it appears to work correctly and lifted the servo.

So I'm 99% sure this is addressed but I'm appreciate if you can let me know what you find.  I'm happy for you to DM me on Discord if you want a more interactive chat.

jwhite, 2024-02-06:
Tested case where filament was somewhere between extruder and encoder.  This worked with the servo in the correct UP position. 

// - DEBUG: Checking for possibility of filament still in extruder gears...
11:37:37 
// - - TRACE: _ensure_safe_extruder_temperature: current_temp=219.99, paused_extruder_temp=None, current_target_temp=220.0, klipper_minimum_temp=180.0, default_extruder_temp=200.0, source=auto
11:37:38 
// - DEBUG: Setting servo to up (filament released) position at angle: 138

Haven't tested a full unload yet.

jwhite, 2024-02-07:
@moggieuk Sorry, still not working.

Reboot printer, MMU_EJECT.  Extruder fights against the servo in 'locked' (move) position and bends/breaks.

10:44:06 (\_/)
         ( *,*)
         (")_(") Happy Hare Ready
10:44:06 Gates: | 0 | 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 |
         Tools: |T0 |T1 |T2 |T3 |T4 |T5 |T6 |T7 |T8 |
         Avail: | ? | ? | ? | ? | ? | ? | ? | ? | ? |
         Selct: ------------------------------------- NOT HOMED
10:44:06 - DEBUG: Job State: READY -> INITIALIZED (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: ready, Position saved: None, z_hop @0.0mm, pause_resume: False, Idle timeout: 1800.00s)
10:44:06 - - TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='initialized' OLD_STATE='ready'
10:44:06 - - TRACE: Disabled runout detection
10:44:06 - DEBUG: Setting servo to move (filament hold) position at angle: 96
10:44:06 - - TRACE: Processing idle_timeout 'printing' event
10:44:07 - - TRACE: Processing idle_timeout 'ready' event
10:44:08 - - TRACE: Processing idle_timeout 'printing' event
10:44:09 - - TRACE: Processing idle_timeout 'ready' event
10:44:18 - - TRACE: Processing idle_timeout 'printing' event
10:44:19 - - TRACE: Processing idle_timeout 'ready' event
10:49:31 MMU is not homed
10:49:38 - DEBUG: Unloading tool unknown
10:49:38 - - TRACE: Running macro: _MMU_PRE_UNLOAD
10:49:38 Attempting to recover filament position...
10:49:38 - - TRACE: (toolhead sensor detects filament)
10:49:38 - - TRACE: (toolhead sensor detects filament)
10:49:38 [T?] < En <<<<<<<<< [Ex << (*) << Nz] LOADED 0.0mm (e:0.0mm)
10:49:38 Unloading filament...
10:49:38 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Unloading' OLD_ACTION='Idle'
10:49:38 - - TRACE: (toolhead sensor detects filament)
10:49:38 [T?] < En <<<<<<<<< [Ex << (*) << Nz] LOADED 0.0mm (e:0.0mm)
10:49:38 - - TRACE: (toolhead sensor detects filament)
10:49:38 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Forming Tip' OLD_ACTION='Unloading'
10:49:38 - - TRACE: _ensure_safe_extruder_temperature: current_temp=219.99, paused_extruder_temp=None, current_target_temp=220.0, klipper_minimum_temp=180.0, default_extruder_temp=200.0, source=auto
10:49:38 - - TRACE: Processing idle_timeout 'printing' event
10:49:39 Forming tip...
10:49:39 - - TRACE: Running macro: _MMU_CUT_TIP_DRAGON
10:49:55 - - TRACE: After tip formation, park_pos reported as: 45.0mm with 2.0mm filament remaining in extruder (extruder moved: 43.0mm, encoder measured 53.4mm)
10:49:55 - - TRACE: (toolhead sensor detects filament)
10:49:55 [T?] < En <<<<<<<<< [Ex << (*) <. Nz] -45.0mm (e:44.9mm)
10:49:55 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Unloading' OLD_ACTION='Forming Tip'
10:49:55 - - TRACE: Running macro: _MMU_POST_FORM_TIP
10:49:55 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Exiting Ext' OLD_ACTION='Unloading'
10:49:55 - DEBUG: Extracting filament from extruder
10:49:55 - - TRACE: _ensure_safe_extruder_temperature: current_temp=220.1, paused_extruder_temp=None, current_target_temp=220.0, klipper_minimum_temp=180.0, default_extruder_temp=200.0, source=auto
10:49:55 - DEBUG: Setting servo to up (filament released) position at angle: 138
10:49:55 - - TRACE: (toolhead sensor detects filament)
10:49:55 - DEBUG: Reverse homing up to 68.0mm to toolhead sensor
10:50:01 - - TRACE: Reverse homing to toolhead sensor. Stepper: 'extruder' did not home after moving -68.0mm (of max -68.0mm), encoder measured 19.7mm (delta 48.3mm). Pos: @-113.0, (66.5mm)
10:50:01 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Unloading' OLD_ACTION='Exiting Ext'
10:50:01 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Idle' OLD_ACTION='Unloading'
10:50:01 An issue with the MMU has been detected whilst out of a print
         Reason: Unload sequence failed: Failed to reach toolhead sensor after moving 68.0mm
10:50:01 - DEBUG: Setting servo to move (filament hold) position at angle: 96
10:50:02 - - TRACE: Processing idle_timeout 'ready' event

This is version : 245d33abe3cf74a6623f89b37f2ffb2b52d90cb5

I suggest this fix.

[image]

Zephyrus67, 2024-03-09:
I have this issue as well. And in my case it is a constant thing.

I have the following settings in mmu_parameters.:
sync_to_extruder: 0
sync_form_tip: 0
sync_feedback_enable: 0

Yet my ERCF 2 now starts every print by setting the servo in POS=MOVE
After each toolchange it also sets the ERCF2 into POS=MOVE

My own fix has been to set MMU_SERVO POS=UP in my nozzle scrubber that is run on every print start in my print_start macro.
That is fixing the problem for the print start. (The MMU_SERVO POS=UP could probably also just be placed in the end of the print_start macro for the printer).

For the toolchange issue I have a MMU_SERVO POS=UP in the _MMU_POST_LOAD macro in mmu_sequences.

At first I used my purge bucket to prime after a toolchange without a purge/wipe tower but I kept getting strange behaviors when starting printing again. Especially clogging warnings without any clogs when checking it manually.

I am now set on parking disabled. filament cutter and purge/wipe block in the slicer.
And the settings I described above.

I DID for a short test try out using FORM_TIP instead of FORM_CUT but I must admit that I did not take any notice of the servor position due to other issues.

Zephyrus67, 2024-03-10:
@moggieuk, @jwhite

I just updated from HH 2.4.?.? to v2.5.0-32-gac57367b

This seems to have fixed the issues.
I also had an issue with z_hop_height_toolchange not being used at all but that also seems fixed now.

Good job.

jwhite, 2024-09-02:
Belived fixed.  Not seeing the issue now.

closed 2024-09-02 by jwhite