moggieuk/Happy-Hare #161: Inconsistent config section naming in multi rail lookup

Xonman, 2024-01-21:
In the `MmuLookupMultiRail` method the code looks for a config section of format `<section name>_<number>` but then attempts to load a config section of format `<section name><number>` with no underscore.

https://github.com/moggieuk/Happy-Hare/blob/202c75d171386cc47ed6da5a199fc460c999d3ee/extras/mmu_toolhead.py#L629-L631

Perhaps something like below is safer:
```python
        section_name = "%s_%s" % (config.get_name(), str(i))
        if not config.has_section(section_name):
            continue
        rail.add_extra_stepper(config.getsection(section_name))
```

moggieuk, 2024-01-21:
Ah, Thank you. "MultiRail" was coded but I've never tested!

Xonman, 2024-01-21:
I figured it was probably an unused path, but spotted it while looking through the rail code and figured it was an easy fix!

closed 2024-02-05 by moggieuk