moggieuk/Happy-Hare #162: [FR] Support for 3dchameleon

stefanpieter, 2024-01-22:
It would be amazing if the 3dchameleon could be used with Happy Hare. This would marry the mechanical simplicity of 3dchameleon with the software capabilities of Happy Hare. 

The ability to use sensor less homing to detect when the filament:
1. Enters the splitter 
2. Tool head 
3. Detects extruder engagement

apoorv1in, 2024-01-27:
[image]
I have a similar request. It is a 4 color MMU with an Homing and filament sensor (gate sensor) on the y splitter. I could not figure out if this setup would be supported by Happy-Hare or will this require a major change or by using just config changes it would work, Or if toolhead sensor is also required with the setup.

moggieuk, 2024-01-28:
WOW! What a super cool design!   If I understand it correctly, you have a "selector" at the top and a "gate" sensor at the Y-split.  No servo.

This is very close to working.  All I can see that would be an issue is that HH currently expects a servo.   I have a virtual servo in the works (which does nothing) but I think your selector would have a "netural" (no hold) position or positions that should correspond to servo "up" (released).   In a way this is similar to the Prusa MMU2 that uses two steppers to select filament and to hold/release it.  I'd love to look at a little more info on this design. Is there anything you could point me to?

apoorv1in, 2024-01-29:
@moggieuk thanks for showing interest in the design.
I have quickly created a git repo with the design CAD files kindly have a look.
Let me know if you have any doubts regarding the design. I am really looking forward to this
Thanks!

https://github.com/apoorv1in/TetraFlow_MMU

akryll, 2024-03-11:
@apoorv1in, hi, is there any progress to support your design in HappyHare?

akryll, 2024-03-11:
> WOW! What a super cool design! If I understand it correctly, you have a "selector" at the top and a "gate" sensor at the Y-split. No servo.
> 
> This is very close to working. All I can see that would be an issue is that HH currently expects a servo. I have a virtual servo in the works (which does nothing) but I think your selector would have a "netural" (no hold) position or positions that should correspond to servo "up" (released). In a way this is similar to the Prusa MMU2 that uses two steppers to select filament and to hold/release it. I'd love to look at a little more info on this design. Is there anything you could point me to?
> 
> As for your toolhead sensor comment -- to work reliably yes you would either need an encoder or a toolhead sensor (or both). The theory of filament loading is very different from the practice where a hundred things can go wrong.
> 
> Put another way... I'd like to build one. If I do, Happy Hare will support it!

You are completely right about the principle of operation. One motor is responsible for feeding the filament and the second is for holding one of them.

The difference from Prush mmu2 is that the selection is made not by a carriage on the shaft, but by means of an eccentric.

Yes, this model has a "neutral". There is a sensor in the splitter and also the ability to connect an encoder.

I have a fully assembled device and am ready to test)
There are nuances regarding engine steps and parking.

apoorv1in, 2024-03-11:
3dchameleon is not an open project so I do not see how it can be supported.. I have taken my git page down..

u3dreal, 2024-03-27:
Parts are available here released by wjsteele ...s ooooo for personal use only 

[https://www.printables.com/de/model/354998-3dchameleon-mk4-upgrade-parts-with-ptfe-grip](url)

glennkaufman, 2024-07-01:
I have successfully (somewhat) integrated my 3dChameleon with the newest HappyHare release. Selector stepper took a minute to dial in, but was just set in position for each gate and calibrated that way. It uses sensorless homing at the the stop position just after Gate 0. Now the part that has me pulling out my hair. Depending on the position gates 0-1 spin clockwise to load, and 2-3 spin counterclockwise on the same stepper to load. Which means all the sequences become reverse when on 2-3. I have my own load macros to get me to the gate / encoder from the buffers with that logic, but I am not looking forward to seeing if I can re-write all the rest. Running the device on a BTT MMB Can. 
[image]
Don't even know where to start. Lol.

OoSTARTERoO, 2024-07-01:
> I have successfully (somewhat) integrated my 3dChameleon with the newest HappyHare release. Selector stepper took a minute to dial in, but was just set in position for each gate and calibrated that way. It uses sensorless homing at the the stop position just after Gate 0. Now the part that has me pulling out my hair. Depending on the position gates 0-1 spin clockwise to load, and 2-3 spin counterclockwise on the same stepper to load. Which means all the sequences become reverse when on 2-3. I have my own load macros to get me to the gate / encoder from the buffers with that logic, but I am not looking forward to seeing if I can re-write all the rest. Running the device on a BTT MMB Can. ![IMG_2592](https://private-user-images.githubusercontent.com/85779251/344537589-75b81097-e8b5-4836-9e6c-682ec0a7754a.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk4MDYxNzMsIm5iZiI6MTcxOTgwNTg3MywicGF0aCI6Ii84NTc3OTI1MS8zNDQ1Mzc1ODktNzViODEwOTctZThiNS00ODM2LTllNmMtNjgyZWMwYTc3NTRhLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcwMVQwMzUxMTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hZDcwNWNkZTA3YmQwZTljNzc1YzY5OTdmZDlkYjIzYzRkMTU2MmRmYTA1YmMyZWI0NDMwZmNhMGMyN2I1NjNjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.0bk2BQR0xKqaURRNMDwbYETk58SIYOgeHH9ufdhMi6g) Don't even know where to start. Lol.

Amazing progress!
But as Moggieuk said, doesn't it require selector "hold" and "release" position?
how did you get it working without the servo configured?

glennkaufman, 2024-07-01:
> Amazing progress! But as Moggieuk said, doesn't it require selector "hold" and "release" position? how did you get it working without the servo configured?

I have servo hooked up but don't use it for anything. It can call out commands to stall day long. It doesn't give me any errors. You could go MMU_MOTORS_OFF if you solely wanted the toolhead extruder once loaded.Or, once the selector stepper is in place, it's just a matter of splitting the load between extruder chameleon gear stepper.

glennkaufman, 2024-07-02:
Can't reverse the _MMU_STEP* macros. So I can't use the call outs to home to sensors on the reverse tools 2-3. Did manage to write a custom load macro through the suggested custom load. So right now just using if-else to call out my moves as positive or negative based on tool selected. Making sure to call out position as the macro goes. Syncs up great with the Klipper screen (Happy Hare). I'm learning the code and logic as I go, so it might take me a minute. Now to reverse this for the unload, and I will get into the numerous other tweaks to make. Keeping all this seperated out in a 3dchameleon.cfg file in Base of mmu. So if anyone else wants a crack at it, let me know.

OoSTARTERoO, 2024-07-02:
> Can't reverse the _MMU_STEP* macros. So I can't use the call outs to home to sensors on the reverse tools 2-3. Did manage to write a custom load macro through the suggested custom load. So right now [...]

I do!, I'm currently building one by myself. your work would be really helpful!

apoorv1in, 2024-07-02:
> Can't reverse the _MMU_STEP* macros. So I can't use the call outs to home to sensors on the reverse tools 2-3. Did manage to write a custom load macro through the suggested custom load. So right now [...]

have a look at this https://klipper.discourse.group/t/switching-several-extruders-on-a-single-driver/2312/2 and https://github.com/Klipper3d/klipper/pull/5210

glennkaufman, 2024-07-02:
> I do!, I'm currently building one by myself. your work would be really helpful!

I'll get some final tweaks done tonight and then find a spot to upload what you might need. What control board will you be using?

glennkaufman, 2024-07-02:
> > Can't reverse the _MMU_STEP* macros. So I can't use the call outs to home to sensors on the reverse tools 2-3. Did manage to write a custom load macro through the suggested custom load. So right now just using if-else to call out my moves as positive or negative based on tool selected. Making sure to call out position as the macro goes. Syncs up great with the Klipper screen (Happy Hare). I'm learning the code and logic as I go, so it might take me a minute. Now to reverse this for the unload, and I will get into the numerous other tweaks to make. Keeping all this seperated out in a 3dchameleon.cfg file in Base of mmu. So if anyone else wants a crack at it, let me know.
> 
> have a look at this https://klipper.discourse.group/t/switching-several-extruders-on-a-single-driver/2312/2 and [Klipper3d/klipper#5210](https://github.com/Klipper3d/klipper/pull/5210)
> 
> SET_EXTRUDER_ROTATION_DISTANCE

I'll dive into this tonight. Thanks for the additional resource. This kind of feels like putting $5000 rims on a $600 car, but I like the challenge. Lol.

u3dreal, 2024-07-08:
Great to see some progress here ... time to take the dust off here. where can i find something to test ?
Also did a preliminary implementation myself but stopped when i realized the tip forming is crucial !

apoorv1in, 2024-07-08:
https://youtu.be/-Z_h8mx2LrQ?si=54lb9BCqUNSUflh0 my implementation in klipper, I am using my own firmware and using serial commands to communicate with mmu

moggieuk, 2026-10-21:
Thanks to everyone who worked on this here. Happy Hare v4 (v4.1 is the current release) supports the 3D Chameleon as a built-in MMU type, so I'm closing this.

Pick "3D Chameleon" as the MMU Type in menuconfig. It uses a dedicated RotarySelector that understands the "opposing gate" design:
- It reverses the gear direction for the gates that load the other way, which was the 0-1 vs 2-3 problem above. No custom load macros are needed.
- It releases filament by selecting the opposing gate.

The defaults (`selector_gate_directions`, `selector_release_gates`, gate parking distance, etc.) are already set for the standard design, so you mainly need to set your board/pins and calibrate.

If your build still needs something v4 doesn't do, please open a new feature request against v4, which is being actively worked on.

closed 2026-10-21 by moggieuk