moggieuk/Happy-Hare #163: Using PrusaSlicer distance from cutter to nozzle must be 15mm shorter than measured

jwhite, 2024-01-25:
I've tested this, both experimentally and thorough measuring the cad.

Because the cutter is so precisely positioned it becomes possible to determine where the cut is and precisely measure the extrusion actually performed before cutting.

On the dragon toolhead the distance between internal nozzle tip and the knife is 59mm.

This must be shortened to 44mm or the knife will not cut.

The retractions performed in the _MMU_CUT_TIP routine are correct, which leads me to believe that there is a strange 15mm retraction occurring from the slicer before the routine is run.  All of my settings are zeroed including ramming.

[image]

jwhite, 2024-01-25:
Actual gcode of a toolchange is here.... not seeing any reason for the 15mm move.

```
;--------------------
; CP TOOLCHANGE START
; toolchange #19
; material : ABS -> ABS
;--------------------
M220 S100
; CP TOOLCHANGE UNLOAD
;WIDTH:0.6
G1 X145.837 Y129.748
SET_PRESSURE_ADVANCE ADVANCE=0
;WIDTH:0.5
G1 F2000
G1 Y129.168 F2400
G4 S0
; Filament-specific end gcode
T4
; Filament gcode

G1 F18000
G1 X145.837 Y129.168
G1 Z2.9 F600
G4 S0
G1 Y129.668
```

jwhite, 2024-01-25:
It's possible I'm getting a fully formed 'nozzle' shaped tip, but I'm lifting it out of the melt pool and not really the nozzle.

moggieuk, 2024-01-27:
Two things:
1. Could I have a clean copy of your cutaway pic for my doc (without dimensions) :-)

2. What you are seeing is a slicer addition and easily fixed.  It was only recently pointed out because I couldn't understand why folks had to artificially shorten their "extruder to nozzle" dimensions to compensate. Now I know...

Set up the slicer like this: https://github.com/moggieuk/Happy-Hare/blob/main/doc/toolchange_movement.md
(its the `retraction length` that you need to zero)

Then make sure the dimensions from you pic are set accurately (+/- 2mm).  You should then ONLY have to tweak the fairly new `toolhead_ooze_reduction` that reduces the theoretical load length to reduce blobs.

Let me know how that works, and I'd love the illustration (I don't have one will all the sensors..)

jwhite, 2024-01-27:
I already have retraction length disabled.

[image]

I'll send you a clean copy of the cutaway.

moggieuk, 2024-01-28:
Thanks for cutaway images!

Looking again at your problem.  The 15mm move isn't in the gcode, so which part of the load/unload process is the 15mm and how do you know it is 15mm? 

The other thing to do is to turn on `log_file_level: 4` in `mmu_parameters.cfg` and restart. This will log every stepper movement that HH makes (outside of the macros) to `mmu.log`. 

Maybe that will help pinpoint.

moggieuk, 2024-01-28:
I had another thought -- you are cutting filament?    How much filament do you leave in the extruder after the cut (that depends on cut macro variables)?   This distance will be detected by HH and the subsequent load will be reduced by this amount.   If you reboot your machine or restart klipper then the first load will "over load" the extruder by this amount becaue it has no idea a bit of filament is left in nozzle..

This seems like the reverse of what you are seeing but thought I'd mention it.

jwhite, 2024-01-28:
I'm using prusaslicer.  All extruders are the same.

I'll try some logging.

moggieuk, 2024-01-29:
What causes you to think that it is ejecting 15mm more than expected?  I'm scratching my head.  Are you sure it isn't the `toolhead_unload_safety_margin`...

`MMU_STATUS SHOWCONFIG=1`
```
Unload Sequence:
- Tip is always formed by '_MMU_FORM_TIP' macro and tip forming extruder current is 120%
- Extruder (optionally synced) unloads by reverse homing a maximum 72.0mm ('toolhead_sensor_to_nozzle + toolhead_unload_safety_margin') to TOOLHEAD SENSOR, then unloads by moving 20.0mm ('toolhead_extruder_to_nozzle - toolhead_sensor_to_nozzle + toolhead_unload_safety_margin') to exit extruder
- Bowden is unloaded with a short 15.0mm ('encoder_move_step_size') validation move before 643.0mm ('calibration_bowden_length - gate_unload_buffer - encoder_move_step_size') fast move
- Filament is stored by homing a maximum of 70.0mm ('gate_homing_max') to ENCODER and parking 23.0mm ('gate_parking_distance') in the gate
```

jwhite, 2024-01-29:
> What causes you to think that it is ejecting 15mm more than expected?

I have measured the distance from the knife to the inside of the nozzle 3 ways (cad, wire with nozzle removed and marked filament.)

The distance is 59mm.

If I set the knife to tooltip distance at 59, when the pushback length is < 43 the blade misses the filament entirely.  Pushback length of 42 allows the blade to cut about 1mm off of the end.  It *should* leave about 16mm in the nozzle (but it doesn't.)

If I mark the blade to tip distance at 44, and the pushback length at 42, I get about 1mm of tip cut.

44mm is about the distance to the cold break from the knife, so it might be pulling a tip from that distance... but that is still contrary to the instructions.

Marking the blade distance at 44 and pushback 42 work well with the dragon hotend I am finding.

moggieuk, 2024-01-29:
Ok. That is interesting and now I understand.  So really this is confined to the _MMU_CUT_TIP macro logic.   Then can't it only be that more filament is left behind in the nozzle than expected -- perhaps breaking in the hot area??

Maybe the macro need a "mini form tip" logic to cleanly separate the filament from the nozzle?

jwhite, 2024-01-29:
> Maybe the macro need a "mini form tip" logic to cleanly separate the filament from the nozzle?

It has the semblance of a cooling move now:
```
    {% if RETRACT_LENGTH > 0 %}
        # Retract to save filament waste, repeat to allow some cooling
        G1 E-{RETRACT_LENGTH} F{extruder_move_speed|float * 60}
        G1 E{RETRACT_LENGTH / 2} F{extruder_move_speed|float * 60} 
        G1 E-{RETRACT_LENGTH / 2} F{extruder_move_speed|float * 60}
    {% endif %}

moggieuk, 2024-01-29:
It's almost as if the filament is "snapping" at the start of the heated area and the filament in the brass nozzle stays there.   15mm is a lot though.  I wonder if retracting more slowly would pull more filament or whether it would just stretch it.

I guess a test would be to just pull the filament out at the space speed as the current retract (with bowden removed in test setup).  Then, manually snip the tip, then extrude incrementally to see the point at which filament extrudes from the nozzle.  I wonder if that would be 68mm or 83mm in your case?

moggieuk, 2024-01-29:
That "cooling move" was something I added (without being the test person) because it helped one user and I thought made sense.  But maybe not good enough.

This is about the only doc so far:
https://github.com/moggieuk/Happy-Hare/blob/main/doc/configuration.md#---toolhead-loading--unloading

jwhite, 2024-01-29:
One other thing to capture is that I am using ABS+ for these tests.  It would be unfortunate if PLA behaved differently.
I have a very nice tip on the part that isn't cut if this is a 'snap.'  The tip is fairly conical with a tiny, but long, string.  

I might mess around with a slow extrude and see if I can get the whole filament out (assuming it is breaking off.)

moggieuk, 2024-01-29:
I'd really appreciate any findings you have.  I still haven't had time to build (assemble) cutting toolhead yet... so many things to do...

reapola, 2024-02-05:
Can confirm the findings above with a rapido2 and CW2 with dual sensor.

I measured the CAD for rapido2 and the distances from the stls and got the following

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/ccmcc/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/ccmcc/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

</head>

<body link="#0563C1" vlink="#954F72">

Measured |  
-- | --
gear entrance to end of cw2 | 26
toolhead sensor to end of cw2 | 15
edge of SB to entrance of rapido | 7.4
rapido top to nozzle | 52
  |  
Calculated |  
toolhead entry to extruder | 15.4
toolhead_sensor_to_nozzle | 74.4
toolhead_extruder_to_nozzle | 85.4
variable_blade_pos | 59.4

</body>

</html>

My current settings of 

variable_retract_length: 35
variable_pushback_length: 33

Causes a clog and as seen from the clogged piece (transparent ABS) fished out, when retracted it pulled partially back with a melted end and when pushed back in it stuck just at the entrance from cold break to heated. Any retract greater than ~35mm misses the filament so it must be splitting somewhere in the heatbreak.

[image]
[image]

reapola, 2024-02-05:
Another clog with the following settings, more left over from the cut but its getting stuck.  

variable_retract_length: 30
variable_pushback_length: 28

[image]

moggieuk, 2024-02-06:
I've come to the conclusion that tip cutting still requires a little bit of tip forming and least to form the basic bullet tip.  The "retraction" in the current cut macro is a straight pull. Normally there is a fast, then slowing first 15mm..  I wonder if it should be a little more sophisticated..?

jwhite, 2024-02-15:
> I've come to the conclusion that tip cutting still requires a little bit of tip forming"

I pulled the dragon out of the printer and played around a bit with filament.  The dragon heatbreak is actually SMALLER than the diameter of the filament, so there will never be a way to pull the filament out cleanly from the nozzle.

When the heat break heats up enough it both exapands and heats/melts the filament at the enterance.  So we really are talking about forming a nice bullet tip at the top  of the heat break tube, not the nozzle.

I'm not sure if this is just a dragon quirk, or all hot ends.

LostFool, 2024-02-18:
I can echo the issues here. I did multiple measurements of my Dragon UHF clone. If I use my measured amounts - nozzle size, It retracts too far and misses the cut. I had to tweak mine and now I have perfect retraction cuts with the tiniest bit left after the cut.

variable_blade_pos: 48 #Hard measure 71.51 nozzle=12.5 logic assumes 59
variable_retract_length: 38

I also use a pushback to try and make sure no swelling issues
variable_pushback_length: 35

I have defeated all of my swap issues (even able to use TPU after a mod I plan to share). Only one last bridge to cross... slow movement to go to cut position and back. Its like it takes a ratio of the current speed.

LostFool, 2024-02-18:
> I've come to the conclusion that tip cutting still requires a little bit of tip forming and least to form the basic bullet tip. The "retraction" in the current cut macro is a straight pull. Normally there is a fast, then slowing first 15mm.. I wonder if it should be a little more sophisticated..?

This fix you pushed out was a huge improvement that allowed me to get smaller "waste" from my cuts. But the discrepancy between measure and actual retract still exists. I haven't dug into why yet.

Anollya, 2024-03-25:
Same here, Dragon SF
variable_blade_pos = 59.5
variable_retract_length  = 39

jwhite, 2024-04-04:
By marking the filament at the extruder instance, then probing the gears (marking), press down and wait till the sensor trips (marking) and lastly the blade (push it in to block the filament, mark) you can obtain reliable measurements using maths and not having to resorting to CAD.

By referencing the TOP of the filament path as zero you remove the 'breaking filament' aspect from the equation.

Lastly, you can seat the filament in a cold (or slightly warm) hotend, mark the extruder entrance, extract by hand, and you can obtain *all* of the correct distances by subtracting the others from this.

You can use a short length of tube protruding from the top of the tube in order to get sub mm accuracy if needed.

I feel like having someone do the exercise (and then doing the calcs in HH to back out the distances) might be an approach that works quickly for all hotends.

davidreact, 2024-06-25:
Thankfully I found this out! I seem to have the same problem. I was using the defaults with a Dragon HF and it has been working ok. I changed to a Bambu clone (which is the same distance as the Dragon HF, give or take one mm here or there) and decided to tune the MMU. If I add the real distance 59mm to the blade_pos when cutting it will retract too much and not allow any cutting. Also when loading I feel that it then loads also too much.

These were the defaults I initially used:
variable_blade_pos              : 37.5		; Distance in mm from internal nozzle tip
variable_retract_length         : 32.5		; (5mm less than 'blade_pos' is a good starting point)

Currently working with this settings
variable_blade_pos              : 44              ; Distance in mm from internal nozzle tip
variable_retract_length         : 32.5		; (5mm less than 'blade_pos' is a good starting point)

Wondering if the issue is to do with: 

variable_simple_tip_forming     : True

This is could be doing some extra retractions and then filament is missing the blade??. I have switched this to False.

LE1ED4523, 2024-06-29:
so what variable_blade_pos for?
use dragon hf and sb toolhead simple_tip_forming enabled,manual push the blade to make a mark on filament but not totally cut it then rewind use mmu_eject and this is what I get
[image]

left to right
1.mark before retract get about 35mm tip. I think it's due to the melt zone so you don't get full 59mm length.
2.variable_blade_pos:30 variable_retract_length:10
mark after retract
console shows
[T5] < (*) <<<<<<<<<< (*) << [Ex << (*) <. Nz] -31.0mm
get about 25mm tip
3.variable_blade_pos:40 variable_retract_length:10
mark after retract
console shows
[T0] < (*) <<<<<<<<<< (*) << [Ex << (*) <. Nz] -41.0mm
get about 25mm tip

no any difference
is that only for measuring length? but I'm not using any encoder

d0m1n1qu3, 2024-07-12:
have a look here . .nice pictures to understand the values
https://github.com/moggieuk/Happy-Hare/wiki/Blobbing-and-Stringing

but for me .. also with a rapido .. i cant find a  config that works good .. got so often a clog that i not fun with the printer /ercf anymore ..  just thinking about to disasemble the ercf an work with onl one color .. :-/

Anollya, 2024-07-12:
It was too stupid for me, I switched to EREC.

https://github.com/kevinakasam/ERCF_Filament_Cutter
Beta 7 should release soon.

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

In v4 the cut-tip macro has a 'simple_tip_forming' option and 'blade_pos' is documented as a value to tune, which covers the shorter-than-measured distance people found here.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk