moggieuk/Happy-Hare #164: Internal error during connect: module 'toolhead' has no attribute 'MoveQueue'

giuseppeferlisi, 2024-01-25:
MMU Hardware Initialization -------------------------------
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 175, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 141, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/pi/klipper/klippy/klippy.py", line 130, in load_object
    self.objects[section] = init_func(config.getsection(section))
  File "/home/pi/klipper/klippy/extras/mmu.py", line 5716, in load_config
    return Mmu(config)
  File "/home/pi/klipper/klippy/extras/mmu.py", line 640, in __init__
    self._setup_mmu_hardware(config)
  File "/home/pi/klipper/klippy/extras/mmu.py", line 652, in _setup_mmu_hardware
    self.mmu_toolhead = MmuToolHead(config, self.homing_extruder)
  File "/home/pi/klipper/klippy/extras/mmu_toolhead.py", line 34, in __init__
    self.move_queue = toolhead.MoveQueue(self) # Happy Hare: Use base class MoveQueue
AttributeError: module 'toolhead' has no attribute 'MoveQueue'

mattb7079, 2024-01-26:
im having the same issue

ningpj, 2024-01-26:
There have been a number of recent changes in klipper and its tool head python object that has created a dependency between klipper and HH versions. You didnt mention what you updated and presume you only updated HH and will also need to update klipper to address this issue.  Once Kevin and the klipper crew finish enhancing the toolhead object, updates will be a little easier in the future.

moggieuk, 2024-01-27:
Ningpj has nailed it.  You just need to make sure Klipper is up-to-date and all will be good.

If klipper changes (and generally it is very rare in the area of overlap) I adjust HH within 24h.  After that change you can't update HH without also updating Klipper..

giuseppeferlisi, 2024-01-27:
ok. thank you. after firmware update (klipper) and a reinstall of HH. it works again

mattb7079, 2024-01-27:
sadly i followed the same steps, did a klipper update via  git pull, reinstalled happy hare and still get the error :/

moggieuk, 2024-01-28:
Both klipper and happy hare need to be up-to-date.   I suspect one is updated and not the other.  To be absolutely sure HH is fully installed run:
```
cd ~/Happy-Hare
./install.sh
```

moggieuk, 2024-02-05:
Closing issue because klipper update fixed it

closed 2024-02-05 by moggieuk