moggieuk/Happy-Hare #168: Pausing after every toolchange after update

zpatten, 2024-01-28:
Assuming this might be related to the most recent commits; I just updated and now things are suddenly pausing after toolchanges.  It seems to be in a weird state as Mainsail reports the print is paused however entering `RESUME` reports it is not paused.  The only way to recover is to call `MMU_PAUSE` and then `RESUME`.

moggieuk, 2024-01-28:
Could you update (force update-manager to refresh) and try again.   Let me know if you still see this.

zpatten, 2024-01-28:
Will do; I'll let you know ASAP.

zpatten, 2024-01-28:
It's still doing it.  :(

In the meantime I've hacked a workaround to do the pause in the post toolchange hook and fire a delayed gcode to handle the resume.  I did notice a bunch of log spam about 'SD card busy' when its doing the pause/delayed resume.

zpatten, 2024-01-28:
After testing the latest commit with my hacky workaround it does appear those SD card busy messages are gone at least; however as mentioned it still seems to get into that paused/busy state and hang without my hacky workaround.

There doesn't appear to be anything in the logs out of the ordinary, certainly no errors.  I suspect you might already know what the problem is since you jumped on it so quicky with that commit (thank you very much by the way, it is appreciated).  

That being said if it would be helpful for me to crank up the logging debug levels just let me know.

jwhite, 2024-01-28:
I am also experiencing this.

thisiscam, 2024-01-29:
I am experiencing this as well. I suspect it could be a bug triggering when the temperatures are different between the swapped filaments. I have reverted to a previously working state and printer is in a long print right now.

@jwhite @zpatten Perhaps could you comment on what's the filament temperatures before and after the swap when this happened for you?

zpatten, 2024-01-29:
I was switching between PLA and PETG; First layer is 220 <-> 245; other layers 240 <-> 260.

thisiscam, 2024-01-29:
> I was switching between PLA and PETG; First layer is 220 <-> 245; other layers 240 <-> 260.

That difference adds to my hypothesis (or wild guess really). 
I briefly looked into the code and also my log, but I wasn't able to spot anything related to temperature changes.

jwhite, 2024-01-29:
@thisiscam I was trying to swap 'cold' at the beginning of my print in order to do MMU_HOME.

zpatten, 2024-01-29:
I can add that what I've observed in my case it makes it past the temperature change.  For me it stops responding after the temp change, filament cut, unload, load; then moves back to the purge block to start purging and hangs there.

zpatten, 2024-01-29:
One other small note; I didn't change anything with my setup; I was printing a file trying to track down a 'timer too close' issue I have intermittently.  I noticed klipper had some changes possibly related to 'timing issues' so I upgraded and then attempted to print the exact same file as before and the pausing after each toolswitch thing started.  I have not tried a print without the temperature change since so it's possible it doesn't hang when using the same material at the same temperature.  I'll try to do a test tonight and see if it still hangs.

zpatten, 2024-01-29:
If it helps anyone else; this is my hacky workaround:

In `_MMU_POST_LOAD` just before the `RESTORE_GCODE_STATE` add:
```
    UPDATE_DELAYED_GCODE ID=Z_MMU_RESUME DURATION=1
    MMU_PAUSE
```

The delayed macro:
```
[delayed_gcode Z_MMU_RESUME]
gcode:
  M117 Z_MMU_RESUME
  RESUME
  MMU_SYNC_GEAR_MOTOR SYNC=1
```

moggieuk, 2024-01-29:
All,  I reverted back the SD card pause logic completely (even though the case that hung was only released for a small window and forcing an update should have fixed it).

I will look at the temperature "pause" issues but need a reproducible case.  The temp restoration logic is surprisingly complex given the number of permutations and possible error cases.

As a side: the `install.sh` script has been enhanced to always look for an update on github when run.  So even though it is often not necessary to run it (update-manager will do), it doesn't hurt and will guarantee that you are up-to-date despite the 24h+ delay that update-manager has is highlighting updates.

zpatten, 2024-01-29:
@moggieuk Awesome, thanks!  After my current print I'll strip out my delayed gcode fix, crank up the log levels and attach the mmu.log.  In my case at least my pausing is happening well after the temperature change and there is nothing in the logs (either klipper or mmu) which indicate it is waiting on anything.

I'll test this before upgrading HH and after (just to be sure your revert fixes it in my case).

zpatten, 2024-01-30:
@moggieuk  I can confirm that the revert fixes things for me at least.  I haven't been able to run on the older version to get a debug log yet.

jwhite, 2024-02-02:
I have the newest version today, and I'm getting forced pauses fairly consistently.

20:20:22 
// probe at 281.244,204.900 is z=8.408894
20:20:25 
// probe at 226.659,290.228 is z=8.468269
20:20:27 
// probe at 226.659,290.228 is z=8.469519
20:20:28 
// probe at 226.659,290.228 is z=8.470144
20:20:29 
// probe at 226.659,290.228 is z=8.470144
20:20:30 
// probe at 226.659,290.228 is z=8.470144
20:20:31 
// Mesh Bed Leveling Complete
20:20:31 
// Bed Mesh state has been saved to profile [default]
// for the current session. The SAVE_CONFIG command will
// update the printer config file and restart the printer.
20:20:38 
// probe: TRIGGERED
20:20:38 
// Bed Mesh state has been saved to profile [auto]
// for the current session. The SAVE_CONFIG command will
// update the printer config file and restart the printer.
20:21:22 
// - - TRACE: _on_print_start(->started)
20:21:22 
// - - TRACE: Enabled runout detection
20:21:22 
// - DEBUG: Job State: PAUSE_LOCKED -> STARTED (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: printing, Position saved: None, z_hop @0.0mm, pause_resume: False, Idle timeout: 72000.00s)
20:21:22 
// - - TRACE: _on_print_start(->printing)
20:21:22 
// MMU initialized ready for print (no filament preloaded)
20:21:22 
// - DEBUG: Job State: STARTED -> PRINTING (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: printing, Position saved: None, z_hop @0.0mm, pause_resume: False, Idle timeout: 72000.00s)
20:21:22 
// - - TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='printing' OLD_STATE='started'
20:21:22 
// - - TRACE: Running macro: _MMU_CLEAR_POSITION
20:21:22 
// - - TRACE: Processing idle_timeout 'ready' event

No amount of resuming starts the print.

jwhite, 2024-02-02:
[image]

jwhite, 2024-02-02:
I have this in my command window :

```
21:07:55 
// Mesh Bed Leveling Complete
21:07:55 
// Bed Mesh state has been saved to profile [default]
// for the current session. The SAVE_CONFIG command will
// update the printer config file and restart the printer.
21:08:02 
// probe: TRIGGERED
21:08:02 
// Bed Mesh state has been saved to profile [auto]
// for the current session. The SAVE_CONFIG command will
// update the printer config file and restart the printer.
21:09:19 
// - - TRACE: _on_print_start(->started)
21:09:19 
// - - TRACE: Enabled runout detection
21:09:19 
// - DEBUG: Job State: PAUSE_LOCKED -> STARTED (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: printing, Position saved: None, z_hop @0.0mm, pause_resume: False, Idle timeout: 72000.00s)
21:09:19 
// - - TRACE: _on_print_start(->printing)
21:09:19 
// MMU initialized ready for print (no filament preloaded)
21:09:19 
// - DEBUG: Job State: STARTED -> PRINTING (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: printing, Position saved: None, z_hop @0.0mm, pause_resume: False, Idle timeout: 72000.00s)
21:09:19 
// - - TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='printing' OLD_STATE='started'
21:09:19 
// - - TRACE: Running macro: _MMU_CLEAR_POSITION
21:09:19 
// - - TRACE: Processing idle_timeout 'ready' event
21:09:20 
$ SET_FILAMENT_SENSOR SENSOR=toolhead_sensor ENABLE=1
```

Pause will not resume for any reason.

However!  In my mmu.log file : 

```
21:01:36 - DEBUG: Tool change initiated with standalone MMU tip forming
21:01:36 Tool change requested: T1
21:01:36 - DEBUG: Loading tool T1...
21:01:36 An issue with the MMU has been detected whilst printer is paused
         **Reason: Gate #1 is empty! Use 'MMU_CHECK_GATE GATE=1' to reset.**
         Occured when changing tool: > T1
21:01:36 Attempting to recover filament position...
21:01:36 - - TRACE: (toolhead sensor does not detect filament)
21:01:36 - DEBUG: Checking for filament in MMU...
21:01:36 - - TRACE: After buzzing gear motor, encoder measured 0.00
21:01:36 - DEBUG: Filament not detected in encoder after buzzing gear motor
21:01:36 - - TRACE: (toolhead sensor does not detect filament)
21:01:36 [T1] > En ......... [Ex .. (t) .. Nz] UNLOADED 70.0mm (e:0.0mm)
21:01:36 - DEBUG: Setting servo to up (filament released) position at angle: 138
21:01:36 - - TRACE: Enabled runout detection
21:01:36 - - TRACE: MMU CLEAR_PAUSE wrapper called
21:01:36 - - TRACE: Running macro: __CLEAR_PAUSE
21:09:19 - - TRACE: _on_print_start(->started)
21:09:19 - - TRACE: Enabled runout detection
21:09:19 - DEBUG: Job State: PAUSE_LOCKED -> STARTED (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: printing, Position saved: None, z_hop @0.0mm, pause_resume: False, Idle timeout: 72000.00s)
21:09:19 - - TRACE: _on_print_start(->printing)
21:09:19 MMU initialized ready for print (no filament preloaded)
21:09:19 - DEBUG: Job State: STARTED -> PRINTING (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: printing, Position saved: None, z_hop @0.0mm, pause_resume: False, Idle timeout: 72000.00s)
21:09:19 - - TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='printing' OLD_STATE='started'
21:09:19 - - TRACE: Running macro: _MMU_CLEAR_POSITION
21:09:19 - - TRACE: Processing idle_timeout 'ready' event
```

In my mmu_parameters.cfg

[image]

Strange that they are so different.

No amount of reset works.

jwhite, 2024-02-02:
Here I 'assisted' the filament into the encoder and loaded it with a T1.

Still, resume fails.

```
21:17:11 
// - - TRACE: Running macro: __RESUME
21:17:11 
// action:resumed
21:17:11 
// - - TRACE: _ensure_safe_extruder_temperature: current_temp=215.0, paused_extruder_temp=215.0, current_target_temp=215.0, klipper_minimum_temp=180.0, default_extruder_temp=200.0, source=pause
21:17:11 
// - - TRACE: Enabled runout detection
21:17:11 
// - DEBUG: Job State: PAUSED -> PRINTING (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: printing, Position saved: pause, z_hop @5.0mm, pause_resume: False, Idle timeout: 1800.00s)
21:17:11 
// - - TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='printing' OLD_STATE='paused'
21:17:11 
// - DEBUG: Restoring toolhead position (X:66.0 Y:316.0 Z:24.9 E:-45.0) after resume
21:17:12 
// - - TRACE: Processing idle_timeout 'ready' event

```​

jwhite, 2024-02-02:
[image]

jwhite, 2024-02-02:
I removed some debug logs here in this thread.  I will re-add them tomorrow after I try some more debugging.

zpatten, 2024-02-02:
@jwhite I had to `MMU_PAUSE` then `RESUME` in order to get it un-paused.  It's like it's in a paused state but doesn't know it, so if you just try `RESUME` it failed for me saying it wasn't paused.  Try issuing the `MMU_PAUSE` before you do the `RESUME` and see if that helps.

jwhite, 2024-02-03:
Here is some logs:

Printer main log - 

```
10:08:57 
// probe at 226.659,290.228 is z=6.794904
10:08:59 
// probe at 226.659,290.228 is z=6.794279
10:09:00 
// probe at 226.659,290.228 is z=6.795529
10:09:00 
// Mesh Bed Leveling Complete
10:09:00 
// Bed Mesh state has been saved to profile [default]
// for the current session. The SAVE_CONFIG command will
// update the printer config file and restart the printer.
10:09:08 
// probe: TRIGGERED
10:09:08 
// Bed Mesh state has been saved to profile [auto]
// for the current session. The SAVE_CONFIG command will
// update the printer config file and restart the printer.
10:10:25 
// - - TRACE: _on_print_start(->started)
10:10:25 
// - - TRACE: Enabled runout detection
10:10:25 
// - DEBUG: Job State: PAUSE_LOCKED -> STARTED (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: printing, Position saved: None, z_hop @0.0mm, pause_resume: False, Idle timeout: 72000.00s)
10:10:25 
// - - TRACE: _on_print_start(->printing)
10:10:25 
// MMU initialized ready for print (initial tool T1 loaded)
10:10:25 
// - DEBUG: Job State: STARTED -> PRINTING (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: printing, Position saved: None, z_hop @0.0mm, pause_resume: False, Idle timeout: 72000.00s)
10:10:25 
// - - TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='printing' OLD_STATE='started'
10:10:25 
// - - TRACE: Running macro: _MMU_CLEAR_POSITION
10:10:25 
// - - TRACE: Processing idle_timeout 'ready' event
```

MMU.log

```
10:01:36 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Homing' OLD_ACTION='Unloading'
10:01:36 - DEBUG: Setting servo to move (filament hold) position at angle: 96
10:01:38 - DEBUG: Selecting tool T1 on gate #1...
10:01:38 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Selecting' OLD_ACTION='Homing'
10:01:38 - - TRACE: Positioning selector
10:01:39 - - TRACE: Setting gear motor rotation distance: 23.387407 (ratio: 1.057722)
10:01:39 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Homing' OLD_ACTION='Selecting'
10:01:39 - DEBUG: Setting servo to up (filament released) position at angle: 138
10:01:39 Tool T1 enabled
10:01:39 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Idle' OLD_ACTION='Homing'
10:01:39 - DEBUG: Loading tool T1...
10:01:39 - - TRACE: Running macro: _MMU_PRE_LOAD
10:01:39 Loading filament...
10:01:39 - - TRACE: (toolhead sensor does not detect filament)
10:01:39 [T1] > En ......... [Ex .. (t) .. Nz] UNLOADED 0.0mm (e:0.0mm)
10:01:39 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Loading' OLD_ACTION='Idle'
10:01:39 - DEBUG: Setting servo to down (filament drive) position at angle: 45
10:01:42 - - TRACE: Initial load into encoder. Stepper: 'gear' moved 70.0mm, encoder measured 54.3mm (delta 15.7mm). Pos: @70.0, (54.3mm)
10:01:42 - - TRACE: (toolhead sensor does not detect filament)
10:01:42 [T1] > En >........ [Ex .. (t) .. Nz] 54.3mm (e:54.3mm)
10:01:42 - DEBUG: Loading bowden tube
10:01:47 - - TRACE: Course loading move into bowden. Stepper: 'gear' moved 716.9mm, encoder measured 728.2mm (delta -11.3mm). Pos: @771.2, (782.5mm)
10:01:47 - - TRACE: (toolhead sensor does not detect filament)
10:01:47 [T1] > En >>>>>>>>> [Ex .. (t) .. Nz] 771.2mm (e:782.5mm)
[... 14 lines cut ...]
10:01:54 Load of 860.1mm filament successful (encoder measured 872.4mm)
10:01:54 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Idle' OLD_ACTION='Loading'
10:01:54 - DEBUG: Setting servo to up (filament released) position at angle: 138
10:01:54 - DEBUG: Spoolman spool_id not set for current gate
10:01:54 - - TRACE: Running macro: _MMU_POST_LOAD
10:01:54 MMU Last Print Statistics:
         1 swaps completed
         14 seconds spent loading (average: 14 seconds)
         57 seconds spent unloading (average: 57 seconds)
         0 seconds spent paused (total pauses: 1)
         
         Gate Statistics:
         #0: Perfect, #1: Perfect, #2: Perfect, #3: Great, #4: Perfect, #5: n/a, #6: n/a, #7: n/a, #8: Good
10:01:54 - - TRACE: Enabled runout detection
10:01:54 - - TRACE: MMU CLEAR_PAUSE wrapper called
10:01:54 - - TRACE: Running macro: __CLEAR_PAUSE
10:10:24 - - TRACE: _on_print_start(->started)
10:10:24 - - TRACE: Enabled runout detection
10:10:24 - DEBUG: Job State: PAUSE_LOCKED -> STARTED (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: printing, Position saved: None, z_hop @0.0mm, pause_resume: False, Idle timeout: 72000.00s)
10:10:24 - - TRACE: _on_print_start(->printing)
10:10:24 MMU initialized ready for print (initial tool T1 loaded)
10:10:24 - DEBUG: Job State: STARTED -> PRINTING (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: printing, Position saved: None, z_hop @0.0mm, pause_resume: False, Idle timeout: 72000.00s)
10:10:24 - - TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='printing' OLD_STATE='started'
10:10:24 - - TRACE: Running macro: _MMU_CLEAR_POSITION
10:10:24 - - TRACE: Processing idle_timeout 'ready' event

jwhite, 2024-02-03:
> @jwhite I had to `MMU_PAUSE` then `RESUME` in order to get it un-paused. It's like it's in a paused state but doesn't know it, so if you just try `RESUME` it failed for me saying it wasn't paused. Try issuing the `MMU_PAUSE` before you do the `RESUME` and see if that helps.

I still can't resume.

Log here :

```
10:23:17 
// - - TRACE: Disabled runout detection
10:23:17 
// - DEBUG: Saving toolhead position (X:66.0 Y:316.0 Z:25.0 E:-87.0) for pause
10:23:17 
// - DEBUG: Lifting toolhead 5.0mm
10:23:17 
// - DEBUG: Job State: PRINTING -> PAUSE_LOCKED (MMU State: Encoder: Disabled, Synced: False, Paused temp: 215.0, Resume to state: printing, Position saved: pause, z_hop @5.0mm, pause_resume: False, Idle timeout: 72000.00s)
10:23:17 
// - - TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='pause_locked' OLD_STATE='printing'
10:23:17 
// - - TRACE: Running macro: PAUSE
10:23:17 
// - - TRACE: MMU PAUSE wrapper called
10:23:17 
// - - TRACE: Running macro: __PAUSE
10:23:17 
// action:paused
10:23:17 
// Attempting to recover filament position...
10:23:17 
// - - TRACE: (toolhead sensor detects filament)
10:23:17 
// - - TRACE: (toolhead sensor detects filament)
10:23:17 
// [T1] > En >>>>>>>>> [Ex >> (*) >> Nz] LOADED 0.0mm (e:0.0mm)
10:23:17 
// - - TRACE: Processing idle_timeout 'printing' event
10:23:18 
// - - TRACE: Processing idle_timeout 'ready' event
10:23:26 
$ RESUME
10:23:26 
// - - TRACE: MMU RESUME wrapper called
10:23:26 
// - DEBUG: Job State: PAUSE_LOCKED -> PAUSED (MMU State: Encoder: Disabled, Synced: False, Paused temp: 215.0, Resume to state: printing, Position saved: pause, z_hop @5.0mm, pause_resume: True, Idle timeout: 72000.00s)
10:23:26 
// - - TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='paused' OLD_STATE='pause_locked'
10:23:26 
// idle_timeout: Timeout set to 1800.00 s
10:23:26 
// - - TRACE: _ensure_safe_extruder_temperature: current_temp=215.05, paused_extruder_temp=215.0, current_target_temp=215.0, klipper_minimum_temp=180.0, default_extruder_temp=200.0, source=pause
10:23:26 
// - - TRACE: Processing idle_timeout 'printing' event
10:23:26 
// - - TRACE: Running macro: __RESUME
10:23:26 
// action:resumed
10:23:26 
// - - TRACE: _ensure_safe_extruder_temperature: current_temp=215.05, paused_extruder_temp=215.0, current_target_temp=215.0, klipper_minimum_temp=180.0, default_extruder_temp=200.0, source=pause
10:23:26 
// - - TRACE: Enabled runout detection
10:23:26 
// - DEBUG: Job State: PAUSED -> PRINTING (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: printing, Position saved: pause, z_hop @5.0mm, pause_resume: False, Idle timeout: 1800.00s)
10:23:26 
// - - TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='printing' OLD_STATE='paused'
10:23:26 
// - DEBUG: Restoring toolhead position (X:66.0 Y:316.0 Z:25.0 E:-88.0) after resume
10:23:27 
// - - TRACE: Processing idle_timeout 'ready' event

moggieuk, 2024-02-05:
Ok. Catching up with "Issues".   Thank you for all the information.  I'm going to ask that you force an update. If you have a fairly recent version of HH then simply:
```
cd ~/Happy-Hare
./install.sh
```
Will also now pull the latest from github.

There were indeed some state issues related to pause/resume but I believe these all to be addressed.  Would you mind verifying.  If you still find issues in the latest code then comment here with `mmu.log` and I'll definitely take a look.

moggieuk, 2024-02-06:
In fixing Issue #179 the interaction of pause / mmu_pause / resume / runout has changed and so I believe this issue would have been addressed.   Therefore I'm going to (optimistically) mark this as "believe fixed"

jwhite, 2024-02-15:
I'm still getting 'stuck' in a paused state.

```
18:48:12 
// - - TRACE: Saved desired extruder temperature: 215.0°C
18:48:12 
// - - TRACE: Extruder heater will be disabled in 10 minutes 0 seconds
18:48:12 
// idle_timeout: Timeout set to 72000.00 s
18:48:12 
// - - TRACE: Disabled runout detection
18:48:12 
// - DEBUG: Asked to save toolhead position for mmu_pause but it is already saved for runout. Ignored
18:48:12 
// - DEBUG: Job State: PRINTING -> PAUSE_LOCKED (MMU State: Encoder: Disabled, Synced: False, Paused temp: 215.0, Resume to state: printing, Position saved: runout, z_hop @2.0mm, pause_resume: True, Idle timeout: 72000.00s)
18:48:13 
// - - TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='pause_locked' OLD_STATE='printing'
18:48:13 
// Attempting to recover filament position...
18:48:13 
// - - TRACE: (toolhead sensor detects filament)
18:48:13 
// - - TRACE: (toolhead sensor detects filament)
18:48:13 
// [T1] > En >>>>>>>>> [Ex >> (*) >> Nz] LOADED 0.0mm (e:2.8mm)
18:48:13 
// - DEBUG: Setting servo to up (filament released) position at angle: 138
18:48:13 
// - - TRACE: Processing idle_timeout 'ready' event
18:48:27 
$ M83
G1 E10 F300
18:48:27 
// - - TRACE: Processing idle_timeout 'printing' event
18:48:28 
$ M83
G1 E10 F300
18:48:28 
$ M83
G1 E10 F300
18:48:29 
$ M83
G1 E10 F300
18:48:29 
$ M83
G1 E10 F300
18:48:30 
$ M83
G1 E10 F300
18:48:40 
// - - TRACE: Processing idle_timeout 'ready' event
18:48:47 
// - - TRACE: Processing idle_timeout 'printing' event
18:48:58 
// - - TRACE: Processing idle_timeout 'ready' event
18:49:02 
// - - TRACE: Processing idle_timeout 'printing' event
18:49:13 
// - - TRACE: Processing idle_timeout 'ready' event
18:49:21 
$ RESUME
18:49:21 
// - - TRACE: MMU RESUME wrapper called
18:49:21 
// idle_timeout: Timeout set to 1800.00 s
18:49:21 
// Enabled extruder heater
18:49:21 
// - - TRACE: _ensure_safe_extruder_temperature: current_temp=214.92, paused_extruder_temp=215.0, current_target_temp=215.0, klipper_minimum_temp=180.0, default_extruder_temp=200.0, source=pause
18:49:21 
// - - TRACE: Processing idle_timeout 'printing' event
18:49:21 
// - DEBUG: Job State: PAUSE_LOCKED -> PAUSED (MMU State: Encoder: Disabled, Synced: False, Paused temp: 215.0, Resume to state: printing, Position saved: runout, z_hop @2.0mm, pause_resume: True, Idle timeout: 1800.00s)
18:49:21 
// - - TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='paused' OLD_STATE='pause_locked'
18:49:21 
// - - TRACE: Running macro: __RESUME
18:49:21 
// action:resumed
18:49:21 
// - - TRACE: _ensure_safe_extruder_temperature: current_temp=214.92, paused_extruder_temp=215.0, current_target_temp=215.0, klipper_minimum_temp=180.0, default_extruder_temp=200.0, source=pause
18:49:21 
// - - TRACE: Enabled runout detection
18:49:21 
// - DEBUG: Job State: PAUSED -> PRINTING (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: printing, Position saved: runout, z_hop @2.0mm, pause_resume: False, Idle timeout: 1800.00s)
18:49:21 
// - - TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='printing' OLD_STATE='paused'
18:49:21 
// - - TRACE: Running macro: _MMU_CLEAR_POSITION
18:49:21 
// - DEBUG: Restoring toolhead position (X:67.0 Y:316.0 Z:24.9 E:17.0) after resume
18:49:24 
// - - TRACE: Processing idle_timeout 'ready' event
18:49:34 
// - - TRACE: MMU RESUME wrapper called
18:49:34 
// Print is not paused. Resume ignored.

​

jwhite, 2024-02-15:
UI shows PAUSE is the current state.

[image]

jwhite, 2024-02-15:
MMU_PAUSE before RESUME as suggested doesn't solve this:

18:56:35 
// - - TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='pause_locked' OLD_STATE='printing'
18:56:35 
// - - TRACE: Running macro: PAUSE
18:56:35 
// - - TRACE: Running macro: PAUSE
18:56:35 
// - - TRACE: MMU PAUSE wrapper called
18:56:35 
// - DEBUG: Asked to save toolhead position for pause but it is already saved for mmu_pause. Ignored
18:56:35 
// - - TRACE: Running macro: __PAUSE
18:56:35 
// action:paused
18:56:35 
// - - TRACE: Processing idle_timeout 'printing' event
18:56:35 
// Attempting to recover filament position...
18:56:35 
// - - TRACE: (toolhead sensor detects filament)
18:56:35 
// - - TRACE: (toolhead sensor detects filament)
18:56:35 
// [T1] > En >>>>>>>>> [Ex >> (*) >> Nz] LOADED 0.0mm (e:0.0mm)
18:56:38 
// - - TRACE: Processing idle_timeout 'ready' event
18:56:50 
$ MMU_RESUME
18:56:51 
// Unknown command:"MMU_RESUME"
18:56:55 
$ RESUME
18:56:55 
// - - TRACE: MMU RESUME wrapper called
18:56:55 
// idle_timeout: Timeout set to 1800.00 s
18:56:55 
// Enabled extruder heater
18:56:55 
// - - TRACE: _ensure_safe_extruder_temperature: current_temp=215.04, paused_extruder_temp=215.0, current_target_temp=215.0, klipper_minimum_temp=180.0, default_extruder_temp=200.0, source=pause
18:56:55 
// - - TRACE: Processing idle_timeout 'printing' event
18:56:55 
// - DEBUG: Job State: PAUSE_LOCKED -> PAUSED (MMU State: Encoder: Disabled, Synced: False, Paused temp: 215.0, Resume to state: printing, Position saved: mmu_pause, z_hop @10.0mm, pause_resume: True, Idle timeout: 1800.00s)
18:56:55 
// - - TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='paused' OLD_STATE='pause_locked'
18:56:55 
// - - TRACE: Running macro: __RESUME
18:56:55 
// action:resumed
18:56:55 
// - - TRACE: _ensure_safe_extruder_temperature: current_temp=215.04, paused_extruder_temp=215.0, current_target_temp=215.0, klipper_minimum_temp=180.0, default_extruder_temp=200.0, source=pause
18:56:55 
// - - TRACE: Enabled runout detection
18:56:55 
// - DEBUG: Job State: PAUSED -> PRINTING (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: printing, Position saved: mmu_pause, z_hop @10.0mm, pause_resume: False, Idle timeout: 1800.00s)
18:56:55 
// - - TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='printing' OLD_STATE='paused'
18:56:55 
// - - TRACE: Running macro: _MMU_CLEAR_POSITION
18:56:55 
// - DEBUG: Restoring toolhead position (X:67.0 Y:316.0 Z:24.9 E:15.0) after resume
18:56:58 
// - - TRACE: Processing idle_timeout 'ready' event
18:57:01 
$ RESUME
18:57:01 
// - - TRACE: MMU RESUME wrapper called
18:57:01 
// Print is not paused. Resume ignored.

​

jwhite, 2024-02-20:
I have found the cause and reproducible steps for my particular case and I have split them out into issue #192 .

I suggest that this be closed as fixed.  @moggieuk

mpsnp, 2024-04-08:
Having the same issue but with a swap of filament in an endless spool mode, I'll check logs tomorrow.

closed 2024-11-29 by github-actions[bot]