moggieuk/Happy-Hare #169: Not able to MMU_CALIBRATE_BOWDEN BOWDEN_LENGTH

f1cs2000, 2024-01-28:
It goes in the wrong direction for MMU_CALIBRATE_BOWDEN BOWDEN_LENGTH, but when I do the test MMU_TEST_MOVE MOVE=100 SPEED=50 it goes towards the extruder and MMU_TEST_MOVE MOVE=-100 SPEED=50 it goes away from the extruder.
[mmu.log](https://github.com/moggieuk/Happy-Hare/files/14075849/mmu.log)

moggieuk, 2024-01-28:
You need to change direction of you gear stepper.  Read here: https://github.com/moggieuk/Happy-Hare/blob/main/doc/hardware_config.md

f1cs2000, 2024-01-28:
I already try that. If I change the direction of the gear stepper, It goes in the wrong direction when I do MMU_TEST_MOVE.

moggieuk, 2024-01-28:
Ok.  There are two ways to calibrate (depending on if you have encoder or not).  Can you tell me what command you are running to do the calibration?

f1cs2000, 2024-01-28:
It's a ERCF v1.1 mod to sb. Was running ok before the mod with the old Happy Hare version.
I run MMU_CALIBRATE_BOWDEN BOWDEN_LENGTH=xxx  I have the Binky encoder.
I just try the manual method and it try to go towards the extruder
Get this error
:55 a.m.
An issue with the MMU has been detected whilst out of a print
Reason: Calibration of bowden length failed. Did not home to gate sensor after moving 530.0mm
8:55 a.m.
Endstop 'mmu_gate' not found

moggieuk, 2024-01-28:
Hmm. Let me triple check. But what is you setting for `gate_homing_endstop:`?   And do you have a gate_endstop configured?

moggieuk, 2024-01-28:
I tested and based on what you say I think you need to set:
```
gate_homing_endstop: encoder
```

Is the direction now fixed?

f1cs2000, 2024-01-28:
[image]
If it's that what you mean ? it was there

moggieuk, 2024-02-05:
Phew!  Ok. glad you figure it out because I was confused as to how this would happen.   

If you don't mind I'm going to close this issue..

closed 2024-02-05 by moggieuk