moggieuk/Happy-Hare #175: Problem on Print Start

MattiaFontanaErrepinet, 2024-02-04:
Hello after doing the whole homing and bed leveling routine. I noticed that he doesn't check the gates that will be used.

In fact it gives me this error: 
`17:03:21 An issue with the MMU has been detected. Print paused.
         Reason: Gate #0 is empty! Use 'MMU_CHECK_GATE GATE=0' to reset.
         Occured when changing tool: Unknown > T0
17:03:21 After fixing the issue, call RESUME to continue printing
         (MMU_UNLOCK can restore temperature)`

The G-Code I'm using inside OrcaSlicer is as follows
`PRINT_START REFERENCED_TOOLS=!referenced_tools! INITIAL_TOOL={initial_tool}
`

         
  And  i have added this to my printer.cfg       
 `[gcode_macro START_PRINT]
description: Called when starting print
gcode:
    {% set REFERENCED_TOOLS = params.REFERENCED_TOOLS|default("")|string %}
    {% set INITIAL_TOOL = params.INITIAL_TOOL|default(0)|int %}

    {% if REFERENCED_TOOLS == "!referenced_tools!" %}
        RESPOND MSG="Happy Hare gcode pre-processor is diabled"
        {% set REFERENCED_TOOLS = INITIAL_TOOL %}
    {% elif REFERENCED_TOOLS == "" %}
        RESPOND MSG="Happy Hare single color print"
        {% set REFERENCED_TOOLS = INITIAL_TOOL %}
    {% endif %}

    :

    MMU_CHECK_GATE TOOLS={REFERENCED_TOOLS}

    MMU_CHANGE_TOOL STANDALONE=1 TOOL={INITIAL_TOOL}    # Optional: load initial tool

    :`

moggieuk, 2024-02-05:
Assuming you have verified that mmu_server in moonraker is enabled, can you try a separate upload to klipper virtual_sd card then initiate the print.  Or better still, upload the file, then open it with your favorite editor and search for PRINT_START to see that it has been expanded.

This does work, but you need to verify what is happening after the gcode is uploaded.

MattiaFontanaErrepinet, 2024-02-05:
They load correctly but it doesn't seem to check

`PRINT_START REFERENCED_TOOLS=0,1 INITIAL_TOOL=0`

moggieuk, 2024-02-06:
That looks correct.  `MMU_CHECK_GATE TOOLS={REFERENCED_TOOLS}` actually initiates the check.

What does `mmu.log` say?  You can attach here..

MattiaFontanaErrepinet, 2024-02-17:
[mmu (2).log](https://github.com/moggieuk/Happy-Hare/files/14319533/mmu.2.log)
Hi this is MMU.log

LostFool, 2024-08-30:
Note the spelling "Happy Hare gcode pre-processor is diabled"
diabled instead of "disabled"

Also I get the same issue after verifying my Print Start, plus start gcode, and mmu_server