moggieuk/Happy-Hare #176: Klipper Crashes on runout event

jwhite, 2024-02-04:
Hash Version : [e](https://github.com/moggieuk/Happy-Hare/commit/e578ca258ec6f8e2a8ebe0437767ed3e1fe71295)

Klipper crashes on run out.  Stack trace here:

Encoder Sensor mmu_encoder: runout event detected, Time 3957.68
action:paused
Unhandled exception during run
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 217, in run
    self.reactor.run()
  File "/home/pi/klipper/klippy/reactor.py", line 292, in run
    g_next.switch()
  File "/home/pi/klipper/klippy/reactor.py", line 340, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/klipper/klippy/reactor.py", line 158, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
                            ^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/klipper/klippy/reactor.py", line 48, in invoke
    res = self.callback(eventtime)
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/klipper/klippy/extras/mmu_encoder.py", line 209, in _runout_event_handler
    self.gcode.run_script("PAUSE\n" + self.runout_gcode + "\nM400")
                          ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
TypeError: can only concatenate str (not "TemplateWrapper") to str
Transition to shutdown state: Unhandled exception during run
Dumping gcode input 0 blocks

jwhite, 2024-02-04:
might need a self.runout_gcode.render() on line 209?

reapola, 2024-02-05:
Same issue, can trigger the crash by just extruding outside of a print.

`Encoder Sensor mmu_encoder: runout event detected, Time 5632.13
action:paused
Unhandled exception during run
Traceback (most recent call last):
  File "/home/reapola/klipper/klippy/klippy.py", line 217, in run
    self.reactor.run()
  File "/home/reapola/klipper/klippy/reactor.py", line 292, in run
    g_next.switch()
  File "/home/reapola/klipper/klippy/reactor.py", line 340, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
  File "/home/reapola/klipper/klippy/reactor.py", line 158, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/home/reapola/klipper/klippy/reactor.py", line 48, in invoke
    res = self.callback(eventtime)
  File "/home/reapola/klipper/klippy/extras/mmu_encoder.py", line 209, in _runout_event_handler
    self.gcode.run_script("PAUSE\n" + self.runout_gcode + "\nM400")
TypeError: can only concatenate str (not "TemplateWrapper") to str
Transition to shutdown state: Unhandled exception during run`

mmu.log

`00:21:01 - DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): 4.3mm
00:21:02 - DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): 3.1mm
00:21:02 - - TRACE: Got sync force feedback update: eventtime=5630.779348606, state=-1
00:21:02 - DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): 1.8mm
00:21:02 - DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): 0.6mm
00:21:02 - DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): 0.6mm
00:21:03 - DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): -0.3mm
00:21:03 - DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): -1.6mm
00:21:04 - DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): -2.8mm
00:21:04 - DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): -4.1mm
00:21:04 - DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): -5.3mm
00:21:04 - DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): -6.6mm
00:21:05 - DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): -7.8mm
00:21:05 - DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): -9.1mm
00:21:05 - DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): -9.4mm
00:21:14 - DEBUG: Klipper disconnected! MMU Shutdown`

jwhite, 2024-02-05:
> Same issue, can trigger the crash by just extruding outside of a print.

My patch (above) will avoid the crash, but I still have strange issues on resume (toolhead doesn't return to position.)

reapola, 2024-02-05:
> Fixed. Very sorry. Just force an update:
> 
> ```
> cd ~/Happy-Hare
> ./install.sh
> ```

Can confirm fixed, caused an intentional runout and the printer paused rather than shutdown.

closed 2024-02-15 by jwhite