moggieuk/Happy-Hare #179: MMU Incorrectly detects clog, then does not return to correct Z height on resume

matthewj301, 2024-02-06:
Hey, having a new issue on 2.42 - Filament clog detection seems to incorrectly determine a clog. I manually extrude some filament to confirm all is good, press RESUME, and the printhead moves to the correct x,y coordinate to continue, but does not return to the print bed and just prints in midair. 

Can post a video if needed tomorrow. I do not see in the mmu log it stating that it is returning to its saved position like I think it should in "_restore_toolhead_position", so possibly there is some sort of state management issue here. 

This is what is reported when I hit RESUME after it thought it detected a clog --> 
04:29:14 - - TRACE: MMU RESUME wrapper called
04:29:14 - - TRACE: Running macro: __RESUME 
04:29:14 - - TRACE: Processing idle_timeout 'printing' event
04:29:14 - - TRACE: Automaticaly detected RESUME (ignored), print_stats=printing, current mmu print_state=printing

I then cancel the print because it is printing above the bed. 

Klippy and MMU logs are attached. 
My MMU configs are here: https://github.com/matthewj301/rat-race/tree/main/mmu

[mmu.log](https://github.com/moggieuk/Happy-Hare/files/14174908/mmu.log)
[klippy (18).log](https://github.com/moggieuk/Happy-Hare/files/14174910/klippy.18.log)

matthewj301, 2024-02-06:
Note, this still occurred after I force updated via the install script per your comments in this issue https://github.com/moggieuk/Happy-Hare/issues/168

J89eu, 2024-02-06:
I've also experienced this one, to re-engage fully I have to resume using a normal resume, then MMU Pause and and MMU resume but this means I have _some_ filament printed 1mm higher on my purge tower which can cause problems later. Video up here showing it but before I understood the issue https://youtu.be/uR6zowkRlB4

moggieuk, 2024-02-06:
Can you tell me what version you are on as reported by in Mainsail, etc... I think that bug was introduced and then removed the same day(?)

[image]

matthewj301, 2024-02-06:
mainsail
v2.9.1

I used the install script last night to force update, and ok, I can try turning down first layer speed, but clog detection only misfires on the first layer right after it loads, not any point afterwards

moggieuk, 2024-02-06:
Watching your video... Nice English accent BTW. My original home country..

matthewj301, 2024-02-06:
So I just reran install.sh, but still on "v2.4.2-47-ga3a23c0e", are you possibly on a different branch to get you to 2.4.2-48? Wondering if that is the main issue here per your comment that the bug was introduced then fixed

Gotcha, nice, I will disable it during purging!

matthewj301, 2024-02-06:
Also, I don't see it restoring toolhead position after this position save in your message above, which is where the issue occurs

02:27:33 - DEBUG: Saving toolhead position (X:154.5 Y:5.0 Z:20.4 E:19.8) for runout

moggieuk, 2024-02-06:
Ok. I think I found the problem....
The runout case is unusual because it (klipper) pauses first and if there is a subsequent mmu error it misses some pause logic...

matthewj301, 2024-02-06:
You're the man, thanks for finding that. Is this something that I can help with in my macros? I'm not doing anything special on pause/resume I believe, leaving it up to Happy Hare to sort that out, but happy to tweak them as needed

moggieuk, 2024-02-06:
Ok. I just pushed an update.  Best to simply run:
```
cd ~/Happy-Hare
./install.sh
```
So you don't have to force update-manager to find it.

Note while in print you can force a runout simply by running `MMU_TEST_RUNOUT`.   Then you can follow that by creating a real error..  Please let me know, the logic in this area of code is making my head hurt - there are so many possible scenarios and orders..

@J89eu your video was very helpful (same issue)... nice English accent BTW (my original home).

matthewj301, 2024-02-06:
Thanks man, running install now, and heating the bed as we speak, so will let you know in 15!

Ya, I can't think of a better way to approach coding for all these possible combination of scenarios :/ 

Any chance RatOS is having a part in this? They do a lot of macro overrides/assumptions to make it easier for new users

moggieuk, 2024-02-06:
RatRig (and Klippain) add an extra layer, but the reason I put in the "safety" logic in HH is to recover even from a bad macro.  Really, so long as pause/resume are paired I layer in outside of these...

J89eu, 2024-02-06:
`happy-hare
v2.4.2-47-ga3a23c0e
mainsail
v2.9.1`

Those are my versions, I do also wonder if I am extruding too fast and triggering the clog macro, I do notice that when my initial purge lines that SuperSlicer puts down to "prep the gates?" are gappy (are those things even needed?), I guess I'll try to figure out what I need to slow down there. Makes sense though. One for later though as I have a big print using the bypass gate going on.

matthewj301, 2024-02-06:
Same result unfortunately, klipper and mmu logs attached. 

Reported Happy-Hare version: v2.4.2-49-g07b19cf2

Taking a page out of @J89eu book and made a video as well (not sure why its a short and not a vid): https://youtube.com/shorts/kEP5Eb7_6XU?feature=share

moggieuk, 2024-02-06:
What!  Ok. now I'm curious...

moggieuk, 2024-02-06:
What is strange is that I'm seeing the restore...
```
16:19:42 - - TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='printing' OLD_STATE='paused'
16:19:42 - DEBUG: Restoring toolhead position (X:154.5 Y:5.0 Z:20.4 E:19.8) after resume
16:19:54 - DEBUG: MMU_CANCEL_PRINT wrapper called
```
just before you cancel.

moggieuk, 2024-02-06:
You are using RatRig's PAUSE/RESUME, right?   Which means that you DONT have `mmu/optional/client_macros.cfg` included. Is that correct?

moggieuk, 2024-02-06:
I see:
```
 DEBUG: Saving toolhead position (X:154.5 Y:5.0 Z:20.4 E:19.8) for runout
 ```
 PAUSE is called which jumps 20mm?
 Now mmu_pause is in correct state
 Then you RESUME which should drop the 20mm, but even if it didn't...
 ```
 16:19:42 - DEBUG: Restoring toolhead position (X:154.5 Y:5.0 Z:20.4 E:19.8) after resume
 ```
 
 What is puzzling is that HH has recorded a Z position of +20mm... why would that be?

matthewj301, 2024-02-06:
Yup, thought that was odd as well that we still have this issue when we see the restored line. So I am actually doing [include mmu/base/*.cfg] which would include optional/client_macros. 

RatOS macros are imported into printer.cfg earlier ([include RatOS/macros.cfg]) so anything from client_macros would override RatOS definition.

And thank you! Just finished the water cooling and AWD mod

moggieuk, 2024-02-06:
... unless somehow your PAUSE is happening first...  Ok. I know what to look for.

matthewj301, 2024-02-06:
Well I don't hit pause, but HH would hit PAUSE first I guess? Z-hop for HH is set to 10

moggieuk, 2024-02-06:
Hmmm, how are we getting to 20mm?
Do you have the mmu/optional/client_macros installed?

matthewj301, 2024-02-06:
Yes, I do have mmu/optional/client_macros because I do [include mmu/*]. This wasn't really intentional and most likely happened during HH1 --> HH2 upgrade, but has been working 

some z-hop variables I'm finding:
z_hop_height_toolchange = 10
z_hop_speed = 30
variable_end_print_park_z_hop = 20

matthewj301, 2024-02-06:
I let it naturally hit the clog detection, used same gcode as last night

matthewj301, 2024-02-06:
Going to try it without mmu client_macros

moggieuk, 2024-02-06:
Ok, so the client_macros.cfg is the only definition of PAUSE/RESUME you have?

There is some strange interaction going on here.   Perhaps if you can explain if what I see in the log makes sense:
1. After startup, you start printing a job with no filament pre-loaded
2. T4 is called, the z -height is at 50mm,  jumping to 60mm for the load, then restores back to 50mm
3. A clog was detected (looks like a real trigger...
4. The z-height is at 20mm which is recorded, 
5. You call resume, and HH returns Z-height to 20mm
6. You cancel.

matthewj301, 2024-02-06:
I just tried without optional/client_macros.cfg, same result. Link here, had to cause 'clog' by unscrewing tension screw lol: https://photos.app.goo.gl/U9VzgFcoEVBrVRU49

1. Correct, filament is in Pika buffer fed into ERCF Gate 4 (t4)
2. Yes, not really sure what is making that 10mm jump, seems Unnecessary but may be an interaction between my safe-z and HH
3. Real trigger yes
4. Yes, 20 is recorded, but if you watch the vid I just linked to, it seems like that is a middle position? Like printer pauses, gets to 20mm, HH sees we have a pause, it thinks we are still on build plate, it records 20 as build height, etc
5. Correct
6. Correct

matthewj301, 2024-02-06:
Let me link you RatOS pause/resume which is what I just used

matthewj301, 2024-02-06:
Pause: https://github.com/Rat-OS/RatOS-configuration/blob/4beae449c3b13b9cac2772047f4e06145197ca3e/macros.cfg#L66
Resume: https://github.com/Rat-OS/RatOS-configuration/blob/4beae449c3b13b9cac2772047f4e06145197ca3e/macros.cfg#L100

moggieuk, 2024-02-06:
Ok. I'm going to have to stare at this but it looks like the RatOS PAUSE is jumping 20mm..
BUT I should be getting  to record the toolhead position BEFORE that runs

matthewj301, 2024-02-06:
Hey, that's progress on solving the mystery, at least we don't have rogue z-hops! But ya, that is very confusing to me as well. Happy-Hare tells RatOS there is an issue (clog) and to pause... Wondering if there is some sort of timing issue going on? Like technically HH calls RatOS pause, then you record toolhead position, but by that point the initial request to RatOS to pause has been fulfilled, leading to the 20mm Z being recorded

moggieuk, 2024-02-06:
I was thinking about timing but I'm the one calling PAUSE and calling `SAVE_GCODE_STATE` (to record position) so it's not clear to me.   This isn't a generic problem because I can't reproduce so I just need to follow through all the logic, including your macros to see if I can figure it out.    I know I fixed @J89eu problem though.

matthewj301, 2024-02-06:
My entire printer config is here if you need https://github.com/matthewj301/rat-race

matthewj301, 2024-02-06:
Possibly someone overriding the others SAVE_GCODE_STATE name?

moggieuk, 2024-02-06:
I think I see it, but need to think careful about the best approach to avoid..

J89eu, 2024-02-06:
If it helps here is my config too and I can also upload mmu.log and klippy.log if it's required
https://github.com/J89eu/CheetahERCF/tree/main

I have not updated just yet

moggieuk, 2024-02-06:
If you have a chance, force an update and try once more.... I don't to say I'm confident this time. lol, but I think so.

matthewj301, 2024-02-06:
Looks promising from initial test, will recreate the above with a legit error (clog) now

matthewj301, 2024-02-06:
Confirmed with a self-produced runout like my above scenarios as well. I think this is fixed! MMU log attached just in case you want to confirm anything you changed in a working situation.

[mmu (3).log](https://github.com/moggieuk/Happy-Hare/files/14185769/mmu.3.log)

Thanks a ton for your help on this!!

moggieuk, 2024-02-06:
Thank you.   This is all because I restructured a section of code because of the new pre-gate sensors (which need to work differently in and out of a print).   Glad it;s working... I'm going to continue some testing tomorrow..

matthewj301, 2024-02-06:
Gotcha, well let me know if you need anyone to test additional stuff! Happy to help. Thanks again!

closed 2024-04-28 by matthewj301