moggieuk/Happy-Hare #180: Loading filament seems to have different behavior as expected when using extruder + toolhead sensors (load errors)

Zephyrus67, 2024-02-08:
It seems like I now and then get an error like this:

"An issue with the MMU has been detected whilst out of a print
Reason: Load sequence failed: Failed to reach toolhead sensor after moving 40.0mm."

I understand that it means that the filament didnt get to the toolhead sensor after moved 40 mm more after it used the initial move from "mmu_calibration_bowden_length" parameter.

I do however NOT understand why I need the "mmu_calibration_bowden_length" parameter at all when I also got the extruder sensor in my printhead.

The calibration of the bowden length between the encoder and the toolhead also appears as "optional" in the hapy hare instructions but when I attempted to silently (and happy) to skip that part all attempts on using the mmu after complained about that I had not finished the calibration and that it was the bowden length calibration I had missed....

SO.......What did I do wrong? DID I do anything wrong.....DO I really need the "mmu_calibration_bowden_length" setup in the vars file?

moggieuk, 2024-02-15:
The "calibration_bowden_length" is used to move the filament quickly to the extruder entrance.  If you have a toolhead sensor this can be a little shorter (so long as your extruder homing max is big enough).

Can you point me to the error in the doc?  But you DO ALWAYS need a "calibration_bowden_length" to control the fast bowden movement which cannot be a homing move because it is too fast -- I run mine at 250mm/s.

Zephyrus67, 2024-02-17:
Sorry for the late response. 
In between I reprinted tophats and the selector because I had some suspicion they had some incorrect sizes. And that seems to have fixed this problem.

I have though run into another thing and that is that I seems to get incorrect clog warnings where the print is stopping.

The error I get is: 
"An issue with the MMU has been detected. Print was already paused.
Reason: A clog has been detected and requires manual intervention".

When I am checking there is no clog though. It DOES to seem to only be when I print with filament from T0. So maybe it is a little tighter than the rest?
The problem is gone when I set servo to UP position.

I am also wondering why the printing process must be done with servo set to MOVE?

Tribute06, 2024-03-07:
Im having an issue where filament loads perfectly from T0, however it fails to reach extruder consistently on the remaining 5 gates. they all load to about 20 mm before extruder then stop. i have even increased max homing in extruder and still nothing.

Zephyrus67, 2024-03-09:
@Tribute06 That was the issue I also basically had.

I realized that the accuracy of my printed parts was not well. After reprinting the parts with more accuracy THIS problem went away. Especially the filament blocks, top hats and the selector changed it from bad to very good.

Tribute06, 2024-03-09:
I will definitely re print the parts and give it a whirl.      Thanks for the guidance bro

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

This was traced to printed part accuracy, and the bowden length is still needed for the fast bowden move (it can be edited directly in mmu_vars.cfg).

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk