moggieuk/Happy-Hare #181: Feature request: Use BTT Smart Filamentsensor instead of Binky Encoder

Rincewind76, 2024-02-12:
Hello all,
I always had my problems in reliability using the old ERCF 1.1 design. It heavily improved with ERCF 2 encoder - but is far away from being stable.

So I am wondering, if it would be possible to use a "commercial" Encoder instead. If I understand correctly the BTT Smart Filament Sensor does exactly what the encoder in the ERCF does: giving regular impulses based on filament movement.

So would it be possible to reconfigure the Happy Hare software to use this encoder instead? I think the only change necessary is the distance between gate exit and encoder position.

thanks for any feedback, bye,
Thomas

moggieuk, 2024-02-15:
I don't know the resolution of the BTT smart sensor but it will work fine today -- just need to set the encoder resolution parameter correctly..

BTW The Binky encoder can be 100% reliable - the ERCFv2 test team have zero problems so it would be worth tracking down you particular issue.

Fragmon, 2025-05-03:
i use the BTT Smart filament sensor as an encoder. Its have a good resolution.

Thats my config for the btt:

[mmu_encoder mmu_encoder]
encoder_pin: ^PF4
encoder_resolution: 4            # This is just a starter value. Overriden by calibrated 'mmu_encoder_resolution' in mmm_vars.cfg
desired_headroom: 5.0            # The clog/runout headroom that MMU attempts to maintain (closest point to triggering runout)
average_samples: 4            # The "damping" effect of last measurement (higher value means slower automatic clog_length reduction)
flowrate_samples: 20            # How many "movements" of the extruder to measure average flowrate over

moggieuk, 2026-10-21:
Thanks for the question, and thanks @Fragmon for confirming it works. To answer it for Happy Hare v4 (v4.1 is the current release): yes, the BTT Smart Filament Sensor can be used as the MMU encoder, because it pulses with filament movement just like Binky. No special support is needed.

In menuconfig:
1. Go to "MMU Features / Additions", enable "Has encoder?" and open "Encoder config".
2. Set Type to "Other" and enter the sensor's pin.
3. Set "Uncalibrated encoder resolution" to a starting value. Fragmon measured about 1.5mm per pulse.
4. Set "Gate endstop to encoder distance" for where the sensor sits in your filament path.

After installing, run `MMU_CALIBRATE_ENCODER` with the filament gripped to measure the exact resolution.

If you find something v4 doesn't handle with this sensor, please open a new feature request against v4, which is being actively worked on.

closed 2026-10-21 by moggieuk