moggieuk/Happy-Hare #182: Z Hop Behaviour

barsamgr, 2024-02-12:
More like a suggestion rather than an issue.
As far as I understand both variable_park_z_hop (mmu_sequence.cfg) and z_hop_height_toolchange (mmu_parameters.cfg) behave the same way, meaning that the lift z in and out of print.  So why not make variable_park_z_hop lift when it is already at park position in order to be lifted when going back in print. This is a change I did already for my purposes because I want the nozzle to drag out of the print but not while going back in.
So this is what I did:
[image]

I moved the move to park command (line 92)  before the z lift logic.

So now if I want a lift in and out of the print I can use z_hop_height_toolchange if I want only going in print I can use variable_park_z_hop.

moggieuk, 2024-02-15:
Hi, yes I can see this is confusing but the the z-hop settings work differently.  There is a little explanation here: https://github.com/moggieuk/Happy-Hare/blob/main/doc/toolchange_movement.md#---z-hop-moves

Really you can choose one or the other depending on your preference.  Not also that I try not to dictate pause/resume macros and these may also have there own z_hop logic.

As you can see in my doc, I set my macro's to 0mm z_hop and just do it with HH because I don't need the movement when not in a print...

moggieuk, 2026-10-21:
Thanks for the suggestion, and sorry it sat so long. Parking and z-hop were completely reworked in Happy Hare v4 (v4.1 is the current release), and you can now get exactly this behaviour with configuration alone, so I'm closing this.

The settings are in `_MMU_SEQUENCE_VARS` (`mmu_macro_vars.cfg`, or menuconfig "Macro Variables" -> "Sequence macros"):
- `variable_park_toolchange` is `x, y, z_hop, z_hop_ramp, retract`. Set its z_hop to 0 so the toolhead moves off the print at print height.
- `variable_pre_unload_position` is `x, y, z_hop`. Give it a z_hop, e.g. `-999, -999, 5`, so the lift happens once the toolhead is already at the park position.

On the way back, Happy Hare travels at that raised height and only drops to print height once it's over the print. That's the "flat going out, lifted coming back" behaviour you were after. If you'd rather break the string with a combined lift and sideways move, look at the `z_hop_ramp` value in the park tuple.

If there's something v4's parking still doesn't let you do, please open a new feature request against v4, which is being actively worked on.

closed 2026-10-21 by moggieuk