moggieuk/Happy-Hare #183: tension sensor not working

knucklebust0, 2024-02-12:
I have build the belay sensor and have it installed and  sync_feedback_enable: 1, I can see the switch status sync_feedback_tension_switch: TRIGGERED when the filament is under tension and it goes to open when not under tension. if i am printing larger prints with a single color the tension switch is triggered but the ercf does not attempt to adjust itself.

moggieuk, 2024-02-15:
Just confirming that you are not using the "Belay" klipper module.  With Happy Hare you simply need to define the pin in `mmu_hardware.cfg`:
```
[mmu_sensors]
sync_feedback_tension_pin: HERE
sync_feedback_compression_pin:
```

fineemb, 2025-04-08:
The belay module has been disabled, and the correct sync_feedback_tension_pin has been configured
and the switch can be triggered normally. However, the feed extruder will not adjust accordingly

nmsmith22389, 2025-04-19:
I’m curious, is there a way to confirm that the adjustments are actually happening? I’ve set up my settings correctly, and the sensor statuses are working, but it’s tough to tell if it’s actually making any changes or how much it’s adjusting in total.

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

Sync feedback was rewritten in v4, and running MMU_SYNC_FEEDBACK with no parameters now gives a status report, so you can see whether it's active.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk