moggieuk/Happy-Hare #186: Printing creates strange title in SpoolMan

jwhite, 2024-02-16:
[image]

Also I'm reasonably sure that the information about spools is already in the gcode, so it shouldn't be prompting me to select.

I am uploading the file then printing from fluidd (so that it is correctly parsed by happy hare.)

moggieuk, 2024-03-10:
Hi! You are prolific at creating issues -- thank you!  You really should get involved in the plumbing and help evolve HH. :-)

Anyway, I'm not sure what you make of this. I'm struggling to parse the issue.  I just finished writing a doc page for spoolman with HH: https://github.com/moggieuk/Happy-Hare/blob/main/doc/spoolman_support.md

jwhite, 2024-06-03:
This seems to have gone away, so marking closed.

closed 2024-06-03 by jwhite