moggieuk/Happy-Hare #187: Avoid Restore Position

barsamgr, 2024-02-17:
Hello.
I would like to have to option to avoid restore position that Happy Hare forces at every tool change. 
The reason is that I have a setup with filament cutter and purge mechanism at very close proximity, so I want to follow a sequence of go to parking position, do the filament change and continue the printing code without going back.  
The next command after the Tx is the Purging command incorporated in Orca slicer where I pass the filament length parameter configured in the flushing volume matrix of the slicer so that I purge specific amount.
The purging is not possible to  be done in the post load process, because I can not pass the length parameter in happy hare (at least I could not find an obvious way to do so).
The sequence as is now it works, in a way, but going back to position and then move again at the parking position in order to do the purging, is unnecessary travel.

So.. Is there a way to avoid the restore position without having to change the code in mmu.py?

barsamgr, 2024-02-24:
This is exactly what I wanted. I had modified mmu.py as a workaround, but I had to change it after each update. Now this is solving my issue. I will get it as soon as I am back home.

moggieuk, 2024-03-10:
Can you comment if the v2.5.0 release fully solves your issue.  I can close this issue out.

moggieuk, 2026-10-21:
Closing this one as it's been done for a long time - you confirmed back in v2.5 that `restore_xy_pos` did what you needed.

In Happy Hare v4 it's still there as `variable_restore_xy_pos` in `mmu_macro_vars.cfg` (or in menuconfig under Macro Variables > Sequence macros > "Restore XY position after toolchange"). It takes `last` (go back to where the toolhead was), `next` (go straight to the next print position) or `none` (leave the toolhead where it is and let the slicer's next move take it there). The starting Z height is always restored.

Also worth knowing for your purge setup: v4 can do the purge itself after a toolchange via `purge_macro` in `mmu.cfg` (`_MMU_PURGE` or `BLOBIFIER`), and the purge amount is taken from the slicer's flush volume matrix (exposed as `printer.mmu.toolchange_purge_volume`), so you don't have to pass the length from the slicer yourself.

If there's something v4 still doesn't do for you, please open a new feature request against v4 - it's being actively worked on.

closed 2026-10-21 by moggieuk