moggieuk/Happy-Hare #190: Toolhead Collision: Unexpected Bed Impact After Successful Tool Changes | Toolhead crashing into bed after Toolchange while printing.

JohnnyBau, 2024-02-18:
After a successful tool change during printing, the toolhead moves approximately 10mm downward into the bed. Then, the movement halts, and after 1-2 seconds, the toolhead moves the same distance upward again. I have ensured that all parameters for Z movement are either turned off or set to zero. I have verified the G-code, and it appears unrelated to the slicer. This movement occurs specifically after a completed tool change.

moggieuk, 2024-02-24:
That shouldn't happen and this is the first report of its kind. Can you attach the 'mmu.log' or 'klippy.log' containing the crash.  There will be info to help diagnose.

JohnnyBau, 2024-02-26:
I appreciate your response. Unfortunately, I'm away from home at the moment. I've noticed that this issue occurs specifically when I use "RESUME" following a failed tool change. Additionally, I encounter the error when purging from Issue #179. Once this error arises, I'm unable to resume the print, regardless of whether I try using HappyHare-Resume or the standard resume feature.

closed 2024-11-29 by github-actions[bot]