moggieuk/Happy-Hare #193: Help me figure out how to turn off any z_hop

deathfrozen, 2024-02-21:
I have already looked through the documentation, including a block with help about z_shop (there are only 2 parameters described there that are responsible for this). I set all the parameters to 0, but still the table goes down and up. This happens at the moment when the new rod is loaded during printing right before the servo goes down to the "servo_down_angle" position.I have already looked through the documentation, including a block with help about z_hop (there are only 2 parameters described there that are responsible for this). I set all the parameters to 0, but still the bed goes down and up. This happens at the moment when the new filament is loaded during printing right before the servo drops to the "servo_down_angle" position.

barsamgr, 2024-02-22:
Have you checked the slicer settings? If you are using SuperSlicer, I have noticed that there is a Z lift just before the tool change command. This is a parameter at the retraction settings of the extruder (Tab Printer Settings).  On the other hand, Orca Slicer does that after the tool change command.

deathfrozen, 2024-02-22:
Yes, I turned off all z_hop options in the slice and i can see from the trajectory in the preview that there is no axis lift (my slicer is Orca Slicer). I can't say if this is due to an update or a configuration change - lately I've changed a lot of things in hardware and updated software. I want to turn off any z_hop, because I suspect that because I suspect that because of this I get uneven layers in height. The model is printed smoothly, but as soon as 2 colors appear in it - the layers become "striped" - each 2 layer it is slightly wider. I also refuse to think that it's the slicer because it happens right during the tool change macro - in the slicer itself, I turned off any tip formation and use Filametrix with the - force_form_tip_standalone option: 1. The only z_hop occurs at the very last moment - the bed is lowered, then the servo disengages the gears and the bed rises back up. As far as I understand, Happy-Hare performs all these actions. In addition, I looked at the gcode that forms the slicer and there is no movement along the Z axis next to the tool change.

barsamgr, 2024-02-23:
Are you sure that not calling _MMU_RESTORE_POSITION solves your problem?  I am asking because I also removed it (for different reason than yours) and I found out that HappyHare forces the restore position in the Python code.  That is why I opened a ticket for this issue that I have.
By the way, the forced restore position can only be seen during printing.

In order to avoid the restore position, I changed the code in mmu.py for the moment until there is a more proper solution for it.

deathfrozen, 2024-02-23:
Yes, it solved my problem. Perhaps the recovery in Python code returns my Z-axis after lifting, rather than GCODE. 

I am referring to the following differences:
[image]
It looks like the position was saved as the upper value, but the command in the macro G1 Z... changes the lower value.

moggieuk, 2024-02-24:
This is an interesting issue, thank you.   So to be clear.  The `_RESTORE_POSITION` does not restore to the correct gcode z-height (assuming the z_hop in HH is set to 0).  If that's the case, can you help be fix?  The corresponding `_SAVE_POSITION` is quite simple but I must be missing something...

deathfrozen, 2024-02-24:
My version of HH is v2.4.2-85-gd299dbaf. Yes, absolutely all the listed parameters are set to 0
[image]
[image]
. I tried to reproduce this bug, I use the macros _MMU_SAVE_POSITION and _MMU_RESTORE_POSITION and I reproduced it a couple of times (i.e. I execute these macros in turn and my Z axis moves each time by the amount of difference between the two parameters as in my screenshot above). I thought that now I would write down several sequences of actions and all the parameters under which this happens. But suddenly it stopped playing. I'm still looking for something that could "fix" these macros. I want to note that both BED MESH and SKEW CORRECTION are used in my klipper, so both could have affected it.

deathfrozen, 2024-02-24:
I don't understand anything and I feel crazy. I've been struggling with these phantom z_hops for 2 days. But when I kind of found the reason and started looking for a way to reproduce, everything was fixed. I restored the _MMU_RESTORE_POSITION macro call in the _MMU_POST_LOAD macro and started printing - z_hop is not happening. All I did was add a string like RESPONSE MSG="{printer.gcode_move}" to the MMU_SAVE_POSITION macro to see what parameters exist there at all. I also removed this line from the macro. The behavior is now completely normal.

moggieuk, 2024-03-10:
Well, that is very strange.  I'm glad it is all working now because I could not see anyway HH would be doing what you describe. I'll leave this issue open for a bit longer but then will close it out..

closed 2024-11-29 by github-actions[bot]