moggieuk/Happy-Hare #195: Cant instaill without moonraker

JesNoFear, 2024-02-23:
no install options to bypass the moonraker check.
If you run Klipper on OctoPrint, you dont have moonraker installed.

moggieuk, 2024-02-24:
Yes, that is an oversight.  Are you able to create a PR?

moggieuk, 2024-02-24:
Also, make sure you read the notes about `_MMU_PRINT_START` and `_MMU_PRINT_END` if printing directly from Octoprint.

JesNoFear, 2024-02-24:
Yeah, I saw the note on adding the _MMU_PRINT_START and _MMU_PRINT_END commands. I am going to have to do a bit of work on my print_start macro as a whole regardless, I am using a Voron TAP, so the current routine is to start tool at 80, get chamber up to temp, then home xyz, heat tool to temp, retract, clean nozzle, drop temp back to 80 with 3 cleans on the way down, home xyz, then qgl, bed mesh if not done in the last 5 prints, then back to temp, purge tool, clean nozzle, then start printing.
Since it will be starting at a state of unloaded, I'll probably be able to cut out half of it, since I won't have oozing from the last print cool down. 

As far as a PR, I'll take a deeper dive into the script today and see what I can come up with, it's not my strongest language, but I am guessing my approach would be to detect the ~/.octoprint folder or the service, to set the klipper config path to ~/ then have it skip the validation check for moonraker. I am also guessing there is a question or setting farther down asking for auto updates using moonraker, so I'll be on the lookout for that as well.

I am still in the process of acquiring all the parts for the ERFC v2, but got the BTT MMB in, and wanted to validate the CAN communication while using the Microfit 2x3 Skirt Connector.

JesNoFear, 2024-02-24:
PR 199

closed 2024-02-25 by JesNoFear