moggieuk/Happy-Hare #196: Option 'step_pin' is not valid in section 'stepper_mmu_gear'

jgbrown54, 2024-02-24:
Trying to set up Happy Hare with Repetier Server. Managed to install using -c and -k options I think. But now I get an error Option 'step_pin' is not valid in section 'stepper_mmu_gear', Not sure how to proceed or what to check.

moggieuk, 2024-02-24:
The installation wasn't successful.  I suspect you didn't get the "-c" directory correct or didn't restart Klipper.  This error occurs when the klipper modules have not been installed (as symlinks) into the klipper/klippy/extras directory

moggieuk, 2024-03-11:
I’m going to close this issue as I’m confident it is simply a miss on the install

closed 2024-03-11 by moggieuk

mlauber, 2024-07-04:
> The installation wasn't successful. I suspect you didn't get the "-c" directory correct or didn't restart Klipper. This error occurs when the klipper modules have not been installed (as symlinks) into the klipper/klippy/extras directory

Ok so how do I fix this?  I had been running happy gare for a few weeks.  I just installed a Beacon sensor and now this error popped up.