moggieuk/Happy-Hare #197: please add an extra runout-sequenz

TTom1088, 2024-02-24:
Hello, I have recently had the problem that when the pre-gate sensors run out, the print head simply stops printing and then simply changes the spool via the endless spool. Since no Prime Tower is approached in this case, the print is usually damaged at this point. Could you integrate your own runout sequence? Just to use a cleaning brush in this case? An activated pause followed by a resume would also be ok.

moggieuk, 2024-02-25:
In the case of a runout HH will perform a tool change. The movement will be defined by the settings in the "sequence" macros. 

Few of things that will be available shortly (v2.5.0 release)
1) The sequence routines have been improved and the ability to insert user defined step is provided.
2) A new `printer.mmu.runout` variable is available that will be 'True' when tool changing as a result of a runout (it will be False on normal too changes).  This will allow for custom logic to handle the lack of a wipe tower.
3) Many folks are moving away from a wipe tower and using a purge system like "Blobifier". This eliminates your problem completely.

So, once v2.5.0 is out it would be a small step to add what you need and I'm happy to consider additions via PRs...

TTom1088, 2024-02-25:
Thanks for the quick reply.

I'll test it with the printer.mmu.runout variable to introduce an extra routine for runout.
I'll stay with Purge Tower. For larger prints, a purge box is not possible with the Voron 2.4.

closed 2024-02-25 by TTom1088