moggieuk/Happy-Hare #202: Internal error on command:"MMU_CHANGE_TOOL"

dominikacri, 2024-02-25:
Get reproducable an exception everytime calling T1 (Tx)  command. 

ERCF2
Easy ERCF Board (SAMD)
SKR Octopus Pro 

No CANBus, boards connected via USB

Default toolhead_extruder_to_nozzle and toolhead_sensor_to_nozzle values. 

[klippy.log](https://github.com/moggieuk/Happy-Hare/files/14395922/klippy.log)
[mmu_config.zip](https://github.com/moggieuk/Happy-Hare/files/14395923/mmu_config.zip)

moggieuk, 2024-02-25:
Thanks. Looking...

moggieuk, 2024-02-25:
Ok. This can happen when the mismatch in "step movement distance" is too great between extruder stepper and gear stepper.

Try one of these three:
1. reduce the microsteps on your extruder - 128 is extremely fine given the 50:10 gear ratio and frankly will reduce torque.  Try 64.
2. Increase the steps on your MMU gear stepper from 16 to 32.
3. Do both of the above.

Let me know.

dominikacri, 2024-02-25:
Top, works now like a charm ! 
Reduced to 32 extruder microsteps which is already more than needed. 
Thanks a lot for your work.

bartoszwas, 2024-02-29:
Had same issue. Weird thing is that my machine was running perfectly for 2-3 months and then suddenly this error... I thought its klipper issue so I did updates on all mcus but didn't help..

Increasing MMU Gear steps to 32 seems to fix it. Extruder (EBB 2209 CAN) was already on 32 steps.

closed 2026-10-21 by moggieuk