moggieuk/Happy-Hare #203: Make Purge Volumes Optionally persistent

barsamgr, 2024-02-25:
I see a lot of functionality in branch variables which I like a lot, especially the handling and loading of purge volumes matrix.  
But I was wondering... Why not be able to save the matrix in mmu_vars.cfg and load that initially or at the absence of PURGE_VOLUMES parameter, so that be able to use the same matrix with other slicers too?
After all the purge volumes does not change frequently or maybe at all ones you have calibrated the numbers for each color transition.

At the moment I use SS and Orca just because SS does not have the ability yet to save a volume matrix.
If I had the option to use multicolor with specific purge volumes with SS would be great.

moggieuk, 2024-02-29:
This is new functionality that will be released very soon and then I'm definitely interested in improving.  A default could certainly be added to `mmu_vars.cfg` but if fully specified it is a big matrix.  BTW This feature works on PS, SS & Orca slicers.

Did you notice that there are fast ways to "populate" the purge matrix? With 1, N, 2*N, and N*N lists.  E.g.
```
> mmu_slicer_tool_map initial_tool=0 purge_volumes=100  DISPLAY=1 DETAIL=1
--------- Slicer MMU Tool Summary ---------
2 color print (Purge volume map loaded)
T0 (Gate 0, unknown, red, 0°C)
T2 (Gate 2, ABS+, 00fe02, 240°C)
Initial Tool: T0
-------------------------------------------
Purge Volume Map:
To -> T0   T1   T2   T3   T4   T5   T6   T7   T8
T0    -   200  200  200  200  200  200  200  200
T1   200   -   200  200  200  200  200  200  200
T2   200  200   -   200  200  200  200  200  200
T3   200  200  200   -   200  200  200  200  200
T4   200  200  200  200   -   200  200  200  200
T5   200  200  200  200  200   -   200  200  200
T6   200  200  200  200  200  200   -   200  200
T7   200  200  200  200  200  200  200   -   200
T8   200  200  200  200  200  200  200  200   - 
```

Or:
```
mmu_slicer_tool_map  purge_volumes=100,110,120,100,110,120,100,110,120
```

etc.

barsamgr, 2024-02-29:
Hello. 
I have noticed all the functionality with the populate, but I want to minimize waste and I have calibrated the amount needed for each transition. There are big difference between color transitions and it is a lot of waste using a specific amount to purge for every color transition.  Also I know that SS and PS has this functionality, BUT it is only available when you use wipe towers, which I don't.  I have setup a purging system that purges the exact amount specified by the matrix.

Another functionality I would love to see (not necessarily in HappyHare), is to be able to to infill first or the first layer just after the tool change and have a variable specifying the amount of filament that will be used for that infill.  This way you can reduce even more the waste of purged filament.

HitLuca, 2024-02-29:
I'm on the same boat as @barsamgr and having this feature implemented would be great. Especially with SS/PS, the purge matrix isn't saved, so for any new sliced file the user would have to pull up a table of values he stored somewhere. This works with purge systems that are not handled by the slicer, of course if someone wants to use a wipe tower he's at the mercy of his slicer of preference and all the related quirks